jr-robotics/robo-gym #13: Failed to make the environment

Shaluols, 2020-11-23:
Thank you for the nice repo, it is really a good step towards the RL simulation. \
I successfully installed the robo-gym (tag/v.0.1.8) under a conda environment and also installed robo-gym-robot-servers under the local python2.7 environment by following the standard installation instructions. \
I am using ubuntu 16.04 + ros kinetic. \
Then I tried the **random_agent_sim.py** example but the program stuck in the line: \
> env = gym.make('NoObstacleNavigationMir100Sim-v0', ip=target_machine_ip, gui=True)

it continuously printing out the following things:
> Starting new Robot Server | Tentative 1
> Starting new Robot Server | Tentative 2
> Starting new Robot Server | Tentative 3
> Starting new Robot Server | Tentative 4
> Starting new Robot Server | Tentative 5
> Starting new Robot Server | Tentative 6
> Starting new Robot Server | Tentative 7
>  ...

Here are the comments run on the terminal:
> cd ~/robo-gym/docs/examples
> conda activate robo-gym 
> start-server-manager (output nothing)
> python random_agent_sim.py 

I tried both **NoObstacleNavigationMir100Sim-v0** and **EndEffectorPositioningUR10** environments, both of them have the same problem. 

Do you have any ideas about this problem?

matteolucchi, 2020-11-23:
Hi @Shaluols ! Thank you for your nice words and for trying out robo-gym! I hope it will help you with tour work! 

This happens when the Client cannot connect to the Server Manager. I have a couple of things you can try out, let's start with this one:

When starting the framework you should have 2 terminal windows open:

1. (_local python 2.7_) where you have to run `$ start-server-manager` 
2. (_robo-gym conda environment_) where you have to run `$ python random_agent_sim.py` 

Please let me know if this solves the issue, otherwise I'll try to figure out what else could be causing this :)

Shaluols, 2020-11-23:
Hi @matteolucchi, thanks for your quick reply! I tried your method and it works! The problem is that I run the _start-server-manager_  under the conda environment. The server manager and the example should be executed in the local python 2.7 and the conda environment respectively.

matteolucchi, 2020-11-23:
Great! I am happy to hear that! If you have any other problem just let us know! :) 

I will close this issue. 

Cheers :)

AKNiloy, 2021-05-06:
Hello good people. So installed the library install and facing this same problem. I am running the server manager on the default python 2.7 base and the py file in the conda activated environment but still getting this "Starting new Robot Server | Tentative 1" thing. My machine is running ubuntu 18.05 LTS with ROS Melodic

matteolucchi, 2021-05-06:
Hello @AKNiloy  :)

After you started the ServerManager could you please run `attach-to-server-manager` and check what's the output there?

Also, I don't know when did you install the robo-gym-server-modules, but we recently released an update with improved logging. Please install the latest version both in the conda environment and in the python 2.7 environment. 

You should find a log file under `robo_gym_server_modules/logs`. 

If you don't know where the robo-gym-server-modules package has been installed you can use `pip show <package name>`

AKNiloy, 2021-05-06:
So Installed everything yesterday so the files should be up to date.

This is what I found with attach-to-server-manager:

2021-05-06 14:57:00,729 - serverManager - INFO - Tentative 7
2021-05-06 14:57:05,732 - serverManager - INFO - Tentative 8
2021-05-06 14:57:10,738 - serverManager - INFO - Tentative 9
2021-05-06 14:57:10,740 - serverManager - ERROR - Could not start Robot Server
NoneType: None
2021-05-06 14:57:10,740 - serverManager - ERROR - Failed to add Robot Server
Traceback (most recent call last):
  File "/home/ak/anaconda3/lib/python3.8/site-packages/robo_gym_server_modules/server_manager/server.py", line 120, in StartNewServer
    assert isinstance(rl_server, int)
AssertionError

AKNiloy, 2021-05-06:
So I found the log:

2021-05-06 14:54:04,379 INFO <PID 24726:MainProcess> serverManager.serve(): Server Manager started at 50100
2021-05-06 14:56:10,504 INFO <PID 24726:MainProcess> serverManager.StartNewServer(): Starting Robot Server...
2021-05-06 14:56:20,680 INFO <PID 24726:MainProcess> serverManager.add_rl_server(): Trying to get state from Robot Server...
2021-05-06 14:56:25,685 INFO <PID 24726:MainProcess> serverManager.add_rl_server(): Tentative 0
2021-05-06 14:56:25,689 DEBUG <PID 24726:MainProcess> serverManager.add_rl_server(): Failed to get state from Robot Server
Traceback (most recent call last):
  File "/home/ak/anaconda3/lib/python3.8/site-packages/robo_gym_server_modules/server_manager/server.py", line 94, in add_rl_server
    assert (len(test_client.get_state())> 1)
  File "/home/ak/anaconda3/lib/python3.8/site-packages/robo_gym_server_modules/robot_server/client.py", line 22, in get_state
    msg = self.robot_server_stub.GetState(robot_server_pb2.Empty(), timeout = 20)
  File "/home/ak/anaconda3/lib/python3.8/site-packages/grpc/_channel.py", line 946, in __call__
    return _end_unary_response_blocking(state, call, False, None)
  File "/home/ak/anaconda3/lib/python3.8/site-packages/grpc/_channel.py", line 849, in _end_unary_response_blocking
    raise _InactiveRpcError(state)
grpc._channel._InactiveRpcError: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.UNAVAILABLE
	details = "failed to connect to all addresses"
	debug_error_string = "{"created":"@1620291385.689603099","description":"Failed to pick subchannel","file":"src/core/ext/filters/client_channel/client_channel.cc","file_line":5419,"referenced_errors":[{"created":"@1620291385.689600564","description":"failed to connect to all addresses","file":"src/core/ext/filters/client_channel/lb_policy/pick_first/pick_first.cc","file_line":397,"grpc_status":14}]}"

This is happening every time apparently

matteolucchi, 2021-05-06:
Ok thank you. 

I need some more info, can you please follow what is described here https://github.com/jr-robotics/robo-gym/issues/26#issuecomment-833292422 and let me know what output do you get?

closed 2021-05-17 by matteolucchi