jr-robotics/robo-gym #14: cannot make the ur10 correctly

ChenyangRan, 2020-11-30:
HI, firstly thanks for your repo, which matches my requirement.
However, I failed to make the UR10 env, which is what I need.
I can make the NoObstacleNavigationMir100Sim-v0, so I think I have do the correct steps the same as #13 .
There is how I try to make the env.
1.Create the terminal windows in python2.7
```
 start-server-manager
```
2. Using the conda env, python3.6
```python
import gym, robo_gym
env = gym.make('EndEffectorPositioningUR10Sim-v0', ip='localhost', gui=True)
```
Then it stucks in a loop.
```
Starting new Robot Server | Tentative 1
Starting new Robot Server | Tentative 2
Starting new Robot Server | Tentative 3
Starting new Robot Server | Tentative 4
Starting new Robot Server | Tentative 5
```
I have tested the roslaunch the ur10sim, but some error happends.
The error is in your another repo. https://github.com/jr-robotics/robo-gym-robot-servers/issues/3.
Could you please help me?

matteolucchi, 2020-11-30:
Hi, this issue is related to a problem with the UR robot-servers in ROS Kinetic, I replied to you in https://github.com/jr-robotics/robo-gym-robot-servers/issues/3 so I will close this one

closed 2020-11-30 by matteolucchi

ChenyangRan, 2020-11-30:
> 
> 
> Hi, this issue is related to a problem with the UR robot-servers in ROS Kinetic, I replied to you in [jr-robotics/robo-gym-robot-servers#3](https://github.com/jr-robotics/robo-gym-robot-servers/issues/3) so I will close this one

It really solved my problem.
Thanks very much.

matteolucchi, 2020-11-30:
Great! I am glad to hear that!