jr-robotics/robo-gym #16: Error occurred while calling the reset function for UR10 tasks

Shaluols, 2020-12-08:
Hi,

I succeed in installing the project and tested the random_agent_sim.py on my computer with ubuntu16.04+Gazebo9.0. The Mir100 related tasks can successfully pass the test. However, I got errors in testing the UR10 related tasks, for example, the 'EndEffectorPositioningUR10Sim-v0' and the 'EndEffectorPositioningUR10DoF5Sim-v0'. The output is: \
> Starting new Robot Server | Tentative 1
> Successfully started Robot Server at localhost:52891
> /home/p285253/miniconda3/envs/robo-gym/lib/python3.6/site-packages/gym/logger.py:30: UserWarning: WARN: Box bound 
> precision lowered by casting to float32
> warnings.warn(colorize('%s: %s'%('WARN', msg % args), 'yellow'))
> Error occurred while calling the reset function. Restarting Robot server ...
> Killing Robot Server at localhost:52891 | Tentative 1
> Successfully killed Robot Server at localhost:52891
> Starting new Robot Server | Tentative 1
> Successfully started Robot Server at localhost:41361
> Error occurred while calling the reset function. Restarting Robot server ...
> Killing Robot Server at localhost:41361 | Tentative 1
> Successfully killed Robot Server at localhost:41361

The Rviz was restarting again and again, but each time before its crash, the arm has some movement. 
Do you have any ideas about this kind of errors?

matteolucchi, 2020-12-10:
Hi! I will try to reproduce this on PC and check whether I get the same behavior. I'll get back to you asap!

matteolucchi, 2020-12-11:
Hi, so if I got it right you are using Ubuntu 16.04 and ROS Kinetic, is that right?

Shaluols, 2020-12-11:
> Hi, so if I got it right you are using Ubuntu 16.04 and ROS Kinetic, is that right?

Yes, I am using Ubuntu 16.04 + Gazebo 9.0 + ROS kinetic.

matteolucchi, 2020-12-11:
Ok, then the issue is given by the fact that the latest version of the UR robot servers is not compatible with ROS Kinetic, similar to what happened in https://github.com/jr-robotics/robo-gym-robot-servers/issues/3 , I copy here the same reply:

> Hi @ChenyangRan and thank you for using robo-gym!
> 
> As it is mentioned at the top of the README of this repository:
> 
> > WARNING: The compatibility of the Universal Robots environments with ROS Kinetic has been maintained until version v0.1.8 included. After this version we integrated a refactored version of the universal_robot repository which is not compatible with ROS Kinetic. We will try to maintain backwards compatibility but if you encounter issues please use v0.1.8 or switch to ROS Melodic.
> 
> The issue here is that we switched to the [melodic-devel-staging branch](https://github.com/ros-industrial/universal_robot/tree/melodic-devel-staging) of [ros-industrial/universal_robot](https://github.com/ros-industrial/universal_robot) and in the new launch file structure it is used the _$dirname_ argument, which is only supported in launch files from ROS Melodic.
> 
> As mentioned in the README you can try to revert back to v0.1.8 of this repository:
> 
> ```
> cd ~/robogym_ws/src/robo-gym-robot-servers
> git checkout v0.1.8
> cd ~/robogym_ws
> catkin build
> ```
> 
> Nevertheless I would suggest you to switch to Ubuntu 18.04 and ROS Melodic + Gazebo 9, especially for the UR Robots we found several improvements in the simulation that were fixed with Gazebo 9.
> 
> Cheers,
> Matteo

Let me know if this fixes it for you :) 

Cheers, 
Matteo

Shaluols, 2020-12-11:
Hi,

Thanks for your reply! My robogym_ws was in the v.0.1.8 tag when this error happened. Then, I reinstalled this server package again and checkout to v.0.1.8 tag. However, the error is still there.

matteolucchi, 2020-12-11:
Hi, I managed to reproduce your error. My previous answer was not complete, also the robo-gym python package needs to be at version v0.1.8. To downgrade the version of the pip package you can use : 
`pip install 'robo-gym==0.1.8' --force-reinstall`

Please let me know if this fixes it.

Shaluols, 2020-12-11:
Hi, I installed the robo-gym from the source, can I change the version to v0.1.8 directly by using git?

matteolucchi, 2020-12-11:
Yes of course, just go to the directory were you cloned the robo-gym repository and  run `git checkout v0.1.8` , after that reinstall with `pip install -e . `

Shaluols, 2020-12-11:
After I checkout to tag v0.1.8, the error disappears! Thank you very much!

closed 2020-12-11 by Shaluols

matteolucchi, 2020-12-11:
You are welcome! I am glad to hear that :)

Alvin-shao, 2023-07-26:
> 不客气！我很高兴听到这个:)

Hello, I have been experiencing this error while using UR5 for simulation. The reset function call has encountered an error. How can I resolve this issue?

Alvin-shao, 2023-07-26:
> 不客气！我很高兴听到这个:)

After I lowered the version of the robot Gym, it showed that there is no UR5 robotic arm available.

Alvin-shao, 2023-07-26:
> 不客气！我很高兴听到这个:)

I am using Ubuntu 20.04