jr-robotics/robo-gym #21: ERROR - Failed to get state from Robot Server

mphamhung, 2021-03-25:
I'm using a docker environment with melodic and ubuntu 18.04. 

When running the following in Python3.7:
```
import gym, robo_gym

env = gym.make('NoObstacleNavigationMir100Sim-v0', ip='localhost', gui=True)
```

I get the following in server-manager
`
2021-03-25 15:11:45,663 - serverManager - INFO - Server Manager started at 50100
2021-03-25 15:26:17,586 - serverManager - ERROR - Failed to get state from Robot Server
Traceback (most recent call last):
  File "/home/user/.local/lib/python2.7/site-packages/robo_gym_server_modules/server_manager/server.py", line 92, in add_rl_server
    assert (len(test_client.get_state())> 1)
  File "/home/user/.local/lib/python2.7/site-packages/robo_gym_server_modules/robot_server/client.py", line 22, in get_state
    msg = self.robot_server_stub.GetState(robot_server_pb2.Empty(), timeout = 20)
  File "/home/user/.local/lib/python2.7/site-packages/grpc/_channel.py", line 923, in __call__
    return _end_unary_response_blocking(state, call, False, None)
  File "/home/user/.local/lib/python2.7/site-packages/grpc/_channel.py", line 826, in _end_unary_response_blocking
    raise _InactiveRpcError(state)
_InactiveRpcError: <_InactiveRpcError of RPC that terminated with:
        status = StatusCode.UNAVAILABLE
        details = "failed to connect to all addresses"
        debug_error_string = "{"created":"@1616685977.586178165","description":"Failed to pick subchannel","file":"src/core/ext/filters/client_channel/client_channel.cc","file_line":5397,"referenced_errors":[{"created":"@1616685977.586165204","description":"failed to connect to all addresses","file":"src/core/ext/filters/client_channel/lb_policy/pick_first/pick_first.cc","file_line":398,"grpc_status":14}]}"
>
`

bungee31, 2021-03-30:
I am having the same issue. Did you manage to solve it?

matteolucchi, 2021-03-30:
Hi, we recently improved the debugging process of robo-gym by including more debug info in the outputs. 

It could be that this error is printed the first time that the client tries to connect to the robot server. The client will try again to connect to the server and if it succeeds everything will works as expected. 
I know this might be misleading because the user will see the error output, I will look into making this more clear. 

Can you please try to repeat the process and let it run for a little longer and then report here the output of the server manager and the output from the environment itself?

bungee31, 2021-03-30:
Hi, 

Thank you for your reply. Apparently the ports where the Robot Server was trying to start were taken. All I had to do what restart the computer and then everything worked as expected! Thank you.

matteolucchi, 2021-03-30:
@bungee31 It's great to hear that!