jr-robotics/robo-gym #24: speed up the training or launch the same env simultaneously

ChenyangRan, 2021-04-13:
Hi, I'm using the 'EndEffectorPositioningUR10Sim-v0', which is from robo_gym_v1.8.0 with ros-kinetic.
It works well, but training an epoch is too time-consuming, is there a way to accelerate the environment or launch the same env simultaneously?
I have tried launch the ur_env simultaneously by 'start-server-manager' twice，and a port is opened. Although the algorithm is still in progress, I don’t know if the two environments are working in parallel.
[image]

Ubunt 16.04
Ros kinetic
Gazebo 9.16

matteolucchi, 2021-04-13:
Hi @ChenyangRan and thank you for using robot-gym! 

You can start the Server Manager once and then call `env.make()` multiple times , with the algorithm that we are using right now we have multiple workers running in parallel and each worker is calling env.make() and the Server Manager spawns a new instance of the env. On a powerful enough machine we run up to 20 workers/environments in parallel. 

Right now we are able to run simulations only up to real time, but we have in our plans to look up solutions to speed up training. 

I am not sure if this answers your question, please let me know if you need further input :)

Matteo

ChenyangRan, 2021-04-13:
> 
> 
> Hi @ChenyangRan and thank you for using robo-gym!
> 
> You can start the Server Manager once and then call `env.make()` multiple times , with the algorithm that we are using right now we have multiple workers running in parallel and each worker is calling env.make() and the Server Manager spawns a new instance of the env. On a powerful enough machine we run up to 20 workers/environments in parallel.
> 
> Right now we are able to run simulations only up to real time, but we have in our plans to look up solutions to speed up training.
> 
> I am not sure if this answers your question, please let me know if you need further input :)
> 
> Matteo

Thanks， maybe I can launch difference parameters to speed my train :)

matteolucchi, 2021-04-13:
Yes, that could be an option. I will close this one for now, feel free to reopen if you need further info.

closed 2021-04-13 by matteolucchi