jr-robotics/robo-gym #27: Failed to add Robot Server

ranjitkathiriya, 2021-06-10:
Hello there,

I am getting an error which is described below. I don't know where I am doing something wrong so I am describing steps that were followed by me.

Steps I followed: 
Created new ROS workspace and Installed this. 
```
`sudo apt-get update && sudo apt-get install apt-utils build-essential psmisc vim-gtk git swig sudo libcppunit-dev python3-catkin-tools python3-rosdep python3-pip python3-rospkg python3-future python3-osrf-pycommon`

export ROBOGYM_WS=~/robogym_ws 
# Set ROS distribution
export ROS_DISTRO=noetic
```

1. robo-gym Side:
Installed and cloned robo-gym to workspace
```
git clone https://github.com/jr-robotics/robo-gym.git
cd robo-gym
pip install -e .
```

2. robo-gym-robot-servers side:
```
mkdir -p $ROBOGYM_WS/src && cd $ROBOGYM_WS/src && git clone https://github.com/jr-robotics/robo-gym-robot-servers.git

sudo sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list' && sudo apt-key adv --keyserver 'hkp://keyserver.ubuntu.com:80' --recv-key C1CF6E31E6BADE8868B172B4F42ED6FBAB17C654

```
Installed Mir_robots:

```
source /opt/ros/$ROS_DISTRO/setup.bash &&\
git clone -b $ROS_DISTRO https://github.com/jr-robotics/mir_robot.git
git clone https://github.com/orocos/orocos_kinematics_dynamics &&\
cd orocos_kinematics_dynamics && git checkout b35c424e77ebc5b7e6f1c5e5c34f8a4666fbf5bc &&\
cd $ROBOGYM_WS &&\
sudo apt-get update &&\
sudo rosdep init && rosdep update &&\
rosdep install --from-paths src -i -y --rosdistro $ROS_DISTRO &&\
catkin init &&\
catkin build &&\
pip3 install robo-gym-server-modules scipy numpy &&\
pip3 install --upgrade protobuf

printf "source /opt/ros/$ROS_DISTRO/setup.bash\nsource $ROBOGYM_WS/devel/setup.bash" >> ~/.bashrc
```
Executed launch file: 

`roslaunch mir100_robot_server sim_robot_server.launch gui:=true`

[image]

Error after executing launch file: 
```
[ INFO] [1623322306.654373634, 29.223000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [robot_description] on the ROS param server.
[ERROR] [1623322306.773747305, 29.223000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/left_wheel_joint
[ERROR] [1623322306.774911554, 29.223000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/right_wheel_joint
[ INFO] [1623322306.780790427, 29.223000000]: Loaded gazebo_ros_control.
[INFO] [1623322307.009624, 29.766000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1623322307.014538, 29.779000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1623322307.019118, 29.787000]: Loading controller: joint_state_controller
[INFO] [1623322307.031260, 29.815000]: Loading controller: mobile_base_controller
[ERROR] [1623322307.038120713, 29.828000000]: Could not load controller 'mobile_base_controller' because controller type 'diff_drive_controller/DiffDriveController' does not exist.
[ERROR] [1623322307.038229137, 29.829000000]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[spawn_urdf-3] process has finished cleanly
log file: /home/kuka-ai/.ros/log/d7e63ff8-c9d9-11eb-b011-7f7aed33a77c/spawn_urdf-3*.log
[ERROR] [1623322308.039793, 32.451000]: Failed to load mobile_base_controller
[INFO] [1623322308.044185, 32.463000]: Controller Spawner: Loaded controllers: joint_state_controller
```

3. robo-gym-server-modules:
    I did `pip install robo-gym-server-modules`, check `start-server-manager && attach-to-server-manager` working okay. 

To execute::

Started: `start-server-manager && attach-to-server-manager`
Run sample file in : python src/robo-gym/docs/examples/random_agent_sim.py

```
2021-06-10 11:33:06,352 - serverManager - INFO - Starting Robot Server...
2021-06-10 11:33:16,461 - serverManager - INFO - Trying to get state from Robot Server...
2021-06-10 11:33:16,461 - serverManager - INFO - Tentative 0 of 20
2021-06-10 11:33:16,483 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:33:21,488 - serverManager - INFO - Tentative 1 of 20
2021-06-10 11:33:21,537 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:33:26,541 - serverManager - INFO - Tentative 2 of 20
2021-06-10 11:33:26,562 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:33:31,568 - serverManager - INFO - Tentative 3 of 20
2021-06-10 11:33:31,589 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:33:36,595 - serverManager - INFO - Tentative 4 of 20
2021-06-10 11:33:36,669 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:33:41,675 - serverManager - INFO - Tentative 5 of 20
2021-06-10 11:33:41,696 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:33:46,701 - serverManager - INFO - Tentative 6 of 20
2021-06-10 11:33:46,723 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:33:51,728 - serverManager - INFO - Tentative 7 of 20
2021-06-10 11:33:51,808 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:33:56,814 - serverManager - INFO - Tentative 8 of 20
2021-06-10 11:33:56,836 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:34:01,841 - serverManager - INFO - Tentative 9 of 20
2021-06-10 11:34:01,863 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:34:06,868 - serverManager - INFO - Tentative 10 of 20
2021-06-10 11:34:06,890 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:34:11,895 - serverManager - INFO - Tentative 11 of 20
2021-06-10 11:34:11,917 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:34:16,921 - serverManager - INFO - Tentative 12 of 20
2021-06-10 11:34:16,942 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:34:21,947 - serverManager - INFO - Tentative 13 of 20
2021-06-10 11:34:21,969 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:34:26,975 - serverManager - INFO - Tentative 14 of 20
2021-06-10 11:34:27,011 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:34:32,016 - serverManager - INFO - Tentative 15 of 20
2021-06-10 11:34:32,038 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:34:37,044 - serverManager - INFO - Tentative 16 of 20
2021-06-10 11:34:37,096 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:34:42,101 - serverManager - INFO - Tentative 17 of 20
2021-06-10 11:34:42,123 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:34:47,129 - serverManager - INFO - Tentative 18 of 20
2021-06-10 11:34:47,164 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:34:52,170 - serverManager - INFO - Tentative 19 of 20
2021-06-10 11:34:52,191 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-10 11:34:57,196 - serverManager - ERROR - Could not start Robot Server
NoneType: None
2021-06-10 11:34:57,197 - serverManager - ERROR - Failed to add Robot Server
Traceback (most recent call last):
  File "/home/kuka-ai/robo-gym-server-modules/robo_gym_server_modules/server_manager/server.py", line 119, in StartNewServer
    assert isinstance(rl_server, int)
AssertionError
2021-06-10 11:34:57,199 - serverManager - INFO - Starting Robot Server...

```

matteolucchi, 2021-06-14:
Hi @ranjitkathiriya ! Sorry for the late reply but I was off for a couple of days. Thank you for using robo-gym and for the detailed issue description, it helps a lot!

It looks like something went wrong during the installation of the ROS dependencies, this sometimes happens if some package repositories are temporarily unavailable. Specifically it looks like the `ros-noetic-diff-drive-controller` package was not installed. 

Please verify that by running:
```
apt list | grep ros-noetic-diff-drive
```
If my prediction is correct you should get no results from this command.

To solve this you can install the missing package with:
```
sudo apt-get install ros-noetic-diff-drive-controller
```
To be on the safe side (maybe also other packages didn't get installed) I would run again the following lines:

```
cd $ROBOGYM_WS &&\
sudo apt-get update &&\
rosdep update &&\
rosdep install --from-paths src -i -y --rosdistro $ROS_DISTRO &&\
catkin build
```

I hope this fixes your issue, please let me know how it goes :)

ranjitkathiriya, 2021-06-14:
Hello @matteolucchi,

Thank you so much for replying and contributing to RL and ROS. This package is really the future of robotic and AI. I have tried the following commands which you gave me then also I am getting this below error.

```
(ros1_envs) kuka-ai@kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:~/rlearning_ws$ apt list | grep ros-noetic-diff-drive

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

ros-noetic-diff-drive-controller-dbgsym/focal 0.18.1-1focal.20210424.000940 amd64
ros-noetic-diff-drive-controller/focal 0.18.1-1focal.20210424.000940 amd64
```

Again I am getting same Error:

```
2021-06-14 16:02:29,309 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:02:34,312 - serverManager - INFO - Tentative 1 of 20
2021-06-14 16:02:34,318 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:02:39,320 - serverManager - INFO - Tentative 2 of 20
2021-06-14 16:02:39,329 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:02:44,332 - serverManager - INFO - Tentative 3 of 20
2021-06-14 16:02:44,339 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:02:49,345 - serverManager - INFO - Tentative 4 of 20
2021-06-14 16:02:49,350 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:02:54,355 - serverManager - INFO - Tentative 5 of 20
2021-06-14 16:02:54,363 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:02:59,368 - serverManager - INFO - Tentative 6 of 20
2021-06-14 16:02:59,375 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:03:04,380 - serverManager - INFO - Tentative 7 of 20
2021-06-14 16:03:04,389 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:03:09,394 - serverManager - INFO - Tentative 8 of 20
2021-06-14 16:03:09,401 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:03:14,407 - serverManager - INFO - Tentative 9 of 20
2021-06-14 16:03:14,413 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:03:19,416 - serverManager - INFO - Tentative 10 of 20
2021-06-14 16:03:19,424 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:03:24,429 - serverManager - INFO - Tentative 11 of 20
2021-06-14 16:03:24,436 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:03:29,441 - serverManager - INFO - Tentative 12 of 20
2021-06-14 16:03:29,449 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:03:34,454 - serverManager - INFO - Tentative 13 of 20
2021-06-14 16:03:34,462 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:03:39,467 - serverManager - INFO - Tentative 14 of 20
2021-06-14 16:03:39,476 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:03:44,482 - serverManager - INFO - Tentative 15 of 20
2021-06-14 16:03:44,487 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:03:49,493 - serverManager - INFO - Tentative 16 of 20
2021-06-14 16:03:49,502 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:03:54,507 - serverManager - INFO - Tentative 17 of 20
2021-06-14 16:03:54,516 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:03:59,520 - serverManager - INFO - Tentative 18 of 20
2021-06-14 16:03:59,529 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:04:04,535 - serverManager - INFO - Tentative 19 of 20
2021-06-14 16:04:04,541 - serverManager - INFO - Waiting 5s before next tentative ...
2021-06-14 16:04:09,547 - serverManager - ERROR - Could not start Robot Server
NoneType: None
2021-06-14 16:04:09,548 - serverManager - ERROR - Failed to add Robot Server
Traceback (most recent call last):
  File "/home/kuka-ai/robo-gym-server-modules/robo_gym_server_modules/server_manager/server.py", line 119, in StartNewServer
    assert isinstance(rl_server, int)
AssertionError
```

matteolucchi, 2021-06-15:
Thank you very much for your kind words!

Could you please provide again the output of 
```
roslaunch mir100_robot_server sim_robot_server.launch gui:=true
```

something is failing when starting the Robot Server, it has to be something related to the ROS installation, hopefully I'll be able to see an error in the output. 

Just to be sure, are you running this on Ubuntu 20.04?

Are you using conda or any virtual environment?

ranjitkathiriya, 2021-06-15:
> roslaunch mir100_robot_server sim_robot_server.launch gui:=true

Output. 

> Just to be sure, are you running this on Ubuntu 20.04?

Yes, I am running this program on Ubuntu 20.04 and has noetic installed on it.

> Are you using conda or any virtual environment?

I am using conda.

```
 WARN] [1623742449.816746008]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame fl_caster_rotation_link at time 20.129000 according to authority unknown_publisher
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame left_wheel_link at time 20.129000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame right_wheel_link at time 20.129000 according to authority unknown_publisher
         at line 278 in [ WARN] [1623742449.816762890]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame fl_caster_wheel_link at time 20.129000 according to authority unknown_publisher
/tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
[ WARN] [1623742449.816780244]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame fr_caster_rotation_link at time 20.129000 according to authority unknown_publisher
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame bl_caster_rotation_link at time 20.159000 according to authority unknown_publisher
         at line [ WARN] [1623742449.816795733]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame fr_caster_wheel_link at time 20.129000 according to authority unknown_publisher
278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame bl_caster_wheel_link at time 20.159000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
[ WARN] [1623742449.816814609]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame left_wheel_link at time 20.129000 according to authority unknown_publisher
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame br_caster_rotation_link at time 20.159000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame br_caster_wheel_link at time 20.159000 according to authority unknown_publisher
[ WARN] [1623742449.816829758]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame right_wheel_link at time 20.129000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame fl_caster_rotation_link at time 20.159000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame fl_caster_wheel_link at time 20.159000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame fr_caster_rotation_link at time 20.159000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame fr_caster_wheel_link at time 20.159000 according to authority unknown_publisher
[ WARN] [1623742449.816879674]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame bl_caster_rotation_link at time 20.159000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame left_wheel_link at time 20.159000 according to authority unknown_publisher
         at line [ WARN] [1623742449.816898009]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame bl_caster_wheel_link at time 20.159000 according to authority unknown_publisher
278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame right_wheel_link at time 20.159000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
[ WARN] [1623742449.816913879]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame br_caster_rotation_link at time 20.159000 according to authority unknown_publisher
Warning: [ WARN] [1623742449.816935260]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame br_caster_wheel_link at time 20.159000 according to authority unknown_publisher
TF_REPEATED_DATA ignoring data with redundant timestamp for frame bl_caster_rotation_link at time 20.242000 according to authority unknown_publisher
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame bl_caster_wheel_link at time 20.242000 according to authority unknown_publisher
         at line 278 in [ WARN] [1623742449.816952373]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame fl_caster_rotation_link at time 20.159000 according to authority unknown_publisher
[... cut ...]
[spawn_urdf-3] process has died [pid 7210, exit code 1, cmd /opt/ros/noetic/lib/gazebo_ros/spawn_model -param robot_description -urdf -model mir -x 0.0 -y 0.0 -Y 0.0 __name:=spawn_urdf __log:=/home/kuka-ai/.ros/log/10cb2222-cdac-11eb-aa71-df36a0c11544/spawn_urdf-3.log].
log file: /home/kuka-ai/.ros/log/10cb2222-cdac-11eb-aa71-df36a0c11544/spawn_urdf-3*.log
[INFO] [1623742450.705031, 28.131000]: Controller Spawner: Loaded controllers: joint_state_controller, mobile_base_controller
[INFO] [1623742450.714067, 28.146000]: Started controllers: joint_state_controller, mobile_base_controller

```

matteolucchi, 2021-06-15:
Ok, at least the previous error is now gone. 

So I think the problem is conda, unfortunately conda and ROS cannot work together out of the box. There are solutions to make that work and we are planning on integrating those in the future, but for the time being here is how to proceed:

The part using ROS needs to be running using the system Python 3. You can still use conda as the python environment in which you run your learning algorithm and the robo-gym package.

To be sure that this is the problem can you try to activate the base environment with:
```
conda activate base
```
Install the required python packages with:
```
pip3 install robo-gym-server-modules scipy numpy &&\
pip3 install --upgrade protobuf
```
and run again 

```
roslaunch mir100_robot_server sim_robot_server.launch gui:=true
```

ranjitkathiriya, 2021-06-15:
These are the common error and warnings?

```
process[robot_server-16]: started with pid [51550]
[ INFO] [1623751634.963680309]: rviz version 1.14.7
[ INFO] [1623751634.963799804]: compiled against Qt version 5.12.8
[ INFO] [1623751634.963824040]: compiled against OGRE version 1.9.0 (Ghadamon)
Traceback (most recent call last):
  File "/home/kuka-ai/rlearning_ws/devel/lib/mir100_robot_server/robot_server.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/kuka-ai/rlearning_ws/src/robo-gym-robot-servers/mir100_robot_server/scripts/robot_server.py", line 3, in <module>
    import grpc
ModuleNotFoundError: No module named 'grpc'
[ INFO] [1623751634.983608571]: Forcing OpenGl version 0.
INFO: cannot create a symlink to latest log directory: [Errno 17] File exists: '/home/kuka-ai/.ros/log/749918d0-cdc1-11eb-aa71-df36a0c11544' -> '/home/kuka-ai/.ros/log/latest'
[robot_server-16] process has died [pid 51550, exit code 1, cmd /home/kuka-ai/rlearning_ws/devel/lib/mir100_robot_server/robot_server.py __name:=robot_server __log:=/home/kuka-ai/.ros/log/749918d0-cdc1-11eb-aa71-df36a0c11544/robot_server-16.log].
log file: /home/kuka-ai/.ros/log/749918d0-cdc1-11eb-aa71-df36a0c11544/robot_server-16*.log
[INFO] [1623751635.104901, 0.000000]: Controller Spawner: Waiting for service controller_manager/load_controller
[ INFO] [1623751635.482536372]: Stereo is NOT SUPPORTED
[ INFO] [1623751635.482667771]: OpenGL device: GeForce GTX 1660 Ti with Max-Q Design/PCIe/SSE2
[ INFO] [1623751635.482689442]: OpenGl version: 4.6 (GLSL 4.6).
[ INFO] [1623751635.645475305]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1623751635.647557637]: waitForService: Service [/gazebo/set_physics_properties] has not been advertised, waiting...
[ INFO] [1623751636.433819196]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ERROR] [1623751636.436811557]: GetModelState: model [mir] does not exist
[ INFO] [1623751636.814172821, 9.910000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1623751636.814277178, 9.910000000]: Starting Laser Plugin (ns = /)
[ INFO] [1623751636.816109569, 9.910000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1623751636.933204850, 9.910000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1623751636.933309397, 9.910000000]: Starting Laser Plugin (ns = /)
[ INFO] [1623751636.935188195, 9.910000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1623751636.945536687, 9.910000000]: <initialOrientationAsReference> is unset, using default value of false to comply with REP 145 (world as orientation reference)
[ INFO] [1623751636.945609935, 9.910000000]: <robotNamespace> set to: //
[ INFO] [1623751636.945624062, 9.910000000]: <topicName> set to: imu_data
[ INFO] [1623751636.945633229, 9.910000000]: <frameName> set to: imu_frame
[ INFO] [1623751636.945658798, 9.910000000]: <updateRateHZ> set to: 50
[ INFO] [1623751636.945684236, 9.910000000]: <gaussianNoise> set to: 2.89e-08
[ INFO] [1623751636.945702149, 9.910000000]: <xyzOffset> set to: 0 0 0
[ INFO] [1623751636.945754017, 9.910000000]: <rpyOffset> set to: 0 -0 0
[ INFO] [1623751636.960902734, 9.910000000]: Physics dynamic reconfigure ready.
[ INFO] [1623751637.013808481, 9.910000000]: Loading gazebo_ros_control plugin
[ INFO] [1623751637.013948645, 9.910000000]: Starting gazebo_ros_control plugin in namespace: /
[ INFO] [1623751637.014829479, 9.910000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [robot_description] on the ROS param server.
[ERROR] [1623751637.140295797, 9.910000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/left_wheel_joint
[ERROR] [1623751637.142207728, 9.910000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/right_wheel_joint
[ INFO] [1623751637.151605144, 9.910000000]: Loaded gazebo_ros_control.
[ WARN] [1623751637.153335201, 9.912000000]: Failed to meet update rate! Took 9.91200000000000081
[ WARN] [1623751637.155017087, 9.914000000]: Failed to meet update rate! Took 9.8889999999999993463
[INFO] [1623751637.221014, 10.011000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1623751637.224416, 10.015000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1623751637.229296, 10.024000]: Loading controller: joint_state_controller
[INFO] [1623751637.248642, 10.074000]: Loading controller: mobile_base_controller
[ INFO] [1623751637.270122771, 10.127000000]: Controller state will be published at 41.2Hz.
[ INFO] [1623751637.272348294, 10.132000000]: Wheel separation will be multiplied by 1.
[ INFO] [1623751637.273710757, 10.135000000]: Left wheel radius will be multiplied by 1.
[ INFO] [1623751637.273789796, 10.136000000]: Right wheel radius will be multiplied by 1.
[ INFO] [1623751637.274475320, 10.137000000]: Velocity rolling window size of 10.
[ INFO] [1623751637.276595254, 10.140000000]: Velocity commands will be considered old if they are older than 0.5s.
[ INFO] [1623751637.277826228, 10.142000000]: Allow mutiple cmd_vel publishers is enabled
[ INFO] [1623751637.281630211, 10.148000000]: Base frame_id set to base_footprint
[ INFO] [1623751637.283358255, 10.151000000]: Odometry frame_id set to odom_comb
[ INFO] [1623751637.285028519, 10.153000000]: Publishing to tf is disabled
[ INFO] [1623751637.300729669, 10.184000000]: Odometry params : wheel separation 0.445208, left wheel radius 0.0625, right wheel radius 0.0625
[ INFO] [1623751637.304289340, 10.188000000]: Adding left wheel with joint name: left_wheel_joint and right wheel with joint name: right_wheel_joint
```

matteolucchi, 2021-06-15:
> Traceback (most recent call last):
  File "/home/kuka-ai/rlearning_ws/devel/lib/mir100_robot_server/robot_server.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/kuka-ai/rlearning_ws/src/robo-gym-robot-servers/mir100_robot_server/scripts/robot_server.py", line 3, in <module>
    import grpc
ModuleNotFoundError: No module named 'grpc'

This is not normal, something went wrong with the grpc installation. Have you tried the steps in the previous reply in the conda base environment?

ranjitkathiriya, 2021-06-15:
Yes, I tried all steps.

matteolucchi, 2021-06-15:
Did you get any error during

> pip3 install robo-gym-server-modules scipy numpy &&\
pip3 install --upgrade protobuf 

?

You can try to install grpc like this
```
python3 -m pip install grpcio
```

to verify that the installation is working start python in the terminal and try to do

```
import grpc
```

ranjitkathiriya, 2021-06-15:
Thank you for replying.

> Did you get any error during ?

Yes, I am not getting any error.

```
(base) kuka-ai@kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:~/rlearning_ws$ pip3 install robo-gym-server-modules scipy numpy &&
> pip3 install --upgrade protobuf
Requirement already satisfied: robo-gym-server-modules in /home/kuka-ai/anaconda3/lib/python3.8/site-packages (0.2.7.6)
Requirement already satisfied: scipy in /home/kuka-ai/anaconda3/lib/python3.8/site-packages (1.5.2)
Requirement already satisfied: numpy in /home/kuka-ai/anaconda3/lib/python3.8/site-packages (1.19.2)
Requirement already satisfied: libtmux in /home/kuka-ai/anaconda3/lib/python3.8/site-packages (from robo-gym-server-modules) (0.9.0)
Requirement already satisfied: grpcio in /home/kuka-ai/anaconda3/lib/python3.8/site-packages (from robo-gym-server-modules) (1.38.0)
Requirement already satisfied: protobuf in /home/kuka-ai/anaconda3/lib/python3.8/site-packages (from robo-gym-server-modules) (3.17.3)
Requirement already satisfied: pyyaml in /home/kuka-ai/anaconda3/lib/python3.8/site-packages (from robo-gym-server-modules) (5.3.1)
Requirement already satisfied: six>=1.5.2 in /home/kuka-ai/anaconda3/lib/python3.8/site-packages (from grpcio->robo-gym-server-modules) (1.15.0)

Requirement already up-to-date: protobuf in /home/kuka-ai/anaconda3/lib/python3.8/site-packages (3.17.3)
Requirement already satisfied, skipping upgrade: six>=1.9 in /home/kuka-ai/anaconda3/lib/python3.8/site-packages (from protobuf) (1.15.0)
```

> to verify that the installation is working start python in the terminal and try to do

```
(base) kuka-ai@kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:~/rlearning_ws$ python3 -m pip install grpcio
Requirement already satisfied: grpcio in /home/kuka-ai/anaconda3/lib/python3.8/site-packages (1.38.0)
Requirement already satisfied: six>=1.5.2 in /home/kuka-ai/anaconda3/lib/python3.8/site-packages (from grpcio) (1.15.0)
(base) kuka-ai@kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:~/rlearning_ws$ python 
Python 3.8.5 (default, Sep  4 2020, 07:30:14) 
[GCC 7.3.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import grpc
>>> 
```

Now I am getting this warning and error is it normal? 

```
14
[ WARN] [1623757428.450180013, 31.408000000]: Failed to meet update rate! Took 0.079000000000000000777
[ WARN] [1623757428.450231941, 31.408000000]: Failed to meet update rate! Took 0.054000000000000006328
[ WARN] [1623757428.451411725, 31.486000000]: Failed to meet update rate! Took 0.077999999999999999889
[ WARN] [1623757428.451487688, 31.486000000]: Failed to meet update rate! Took 0.05300000000000000544
[ INFO] [1623757428.677861914, 31.486000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1623757428.677915916, 31.486000000]: Starting Laser Plugin (ns = /)
[ INFO] [1623757428.679152076, 31.486000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1623757428.796911994, 31.486000000]: Laser Plugin: Using the 'robotNamespace' param: '/'
[ INFO] [1623757428.796969623, 31.486000000]: Starting Laser Plugin (ns = /)
[ INFO] [1623757428.799169499, 31.486000000]: Laser Plugin (ns = /)  <tf_prefix_>, set to ""
[ INFO] [1623757428.807372122, 31.486000000]: <initialOrientationAsReference> is unset, using default value of false to comply with REP 145 (world as orientation reference)
[ INFO] [1623757428.807440891, 31.486000000]: <robotNamespace> set to: //
[ INFO] [1623757428.807456571, 31.486000000]: <topicName> set to: imu_data
[ INFO] [1623757428.807472040, 31.486000000]: <frameName> set to: imu_frame
[ INFO] [1623757428.807501716, 31.486000000]: <updateRateHZ> set to: 50
[ INFO] [1623757428.807514881, 31.486000000]: <gaussianNoise> set to: 2.89e-08
[ INFO] [1623757428.807529659, 31.486000000]: <xyzOffset> set to: 0 0 0
[ INFO] [1623757428.807553985, 31.486000000]: <rpyOffset> set to: 0 -0 0
[ INFO] [1623757428.860675309, 31.486000000]: Loading gazebo_ros_control plugin
[ INFO] [1623757428.860872460, 31.486000000]: Starting gazebo_ros_control plugin in namespace: /
[ INFO] [1623757428.863371741, 31.486000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [robot_description] on the ROS param server.
[ERROR] [1623757428.986887078, 31.486000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/left_wheel_joint
[ERROR] [1623757428.988162903, 31.486000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/right_wheel_joint
```

matteolucchi, 2021-06-15:
This

> ERROR] [1623757535.641098273, 20.665000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/left_wheel_joint
[ERROR] [1623757535.643575092, 20.665000000]: No p gain specified for pid.  Namespace: /gazebo_ros_control/pid_gains/right_wheel_joint

and these 

> [ WARN] [1623757604.852068532, 24.017000000]: Failed to meet update rate! Took 0.096000000000000001998
[ WARN] [1623757604.852122384, 24.020000000]: Failed to meet update rate! Took 0.07100000000000000755
[ WARN] [1623757604.853273413, 24.074000000]: Failed to meet update rate! Took 0.057000000000000002054
[ WARN] [1623757604.854492211, 24.148000000]: Failed to meet update rate! Took 0.081000000000000002554
[ WARN] [1623757604.854563325, 24.168000000]: Failed to meet update rate! Took 0.075999999999999998113
[ WARN] [1623757604.855709535, 24.270000000]: Failed to meet update rate! Took 0.12200000000000001121
[ WARN] [1623757604.855770350, 24.270000000]: Failed to meet update rate! Took 0.097000000000000002887
[ WARN] [1623757604.856896813, 24.389000000]: Failed to meet update rate! Took 0.11900000000000000855
[ WARN] [1623757604.856983516, 24.389000000]: Failed to meet update rate! Took 0.094000000000000000222

are ok. 

The problem now is here:

> [ERROR] [1623757535.011737738, 11.774000000]: GetModelState: model [mir] does not exist
[ERROR] [1623757535.014937, 0.000000]: Spawn service failed. Exiting.

Please try to build again your workspace with `catkin build` and to source it 
```
source /opt/ros/$ROS_DISTRO/setup.bash
source $ROBOGYM_WS/devel/setup.bash
```

and now try again the roslaunch. 

Just to be sure have you completed this stage of the installation?

> 6. Add the sourcing of ROS and the ROS workspace to your .bashrc file:
printf "source /opt/ros/$ROS_DISTRO/setup.bash\nsource $ROBOGYM_WS/devel/setup.bash" >> ~/.bashrc

ranjitkathiriya, 2021-06-15:
Yes! I followed the following steps.

> Catkin build 
```
(base) kuka-ai@kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:~/rlearning_ws$ catkin build
---------------------------------------------------------------
Profile:                     default
Extending:          [cached] /opt/ros/noetic
Workspace:                   /home/kuka-ai/rlearning_ws
---------------------------------------------------------------
Build Space:        [exists] /home/kuka-ai/rlearning_ws/build
Devel Space:        [exists] /home/kuka-ai/rlearning_ws/devel
Install Space:      [unused] /home/kuka-ai/rlearning_ws/install
Log Space:          [exists] /home/kuka-ai/rlearning_ws/logs
Source Space:       [exists] /home/kuka-ai/rlearning_ws/src
DESTDIR:            [unused] None
---------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
---------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------------
Workspace configuration appears valid.
---------------------------------------------------------------
[build] Found '13' packages in 0.0 seconds.                                                        
[build] Package table is up to date.                                                               
Starting  >>> mir_actions                                                                          
Starting  >>> mir_description                                                                      
Starting  >>> mir_dwb_critics                                                                      
Starting  >>> mir_msgs                                                                             
Starting  >>> orocos_kdl                                                                           
Starting  >>> sdc21x0                                                                              
Finished  <<< mir_description                           [ 0.1 seconds ]                            
Finished  <<< mir_dwb_critics                           [ 0.5 seconds ]                            
Finished  <<< mir_actions                               [ 0.5 seconds ]                            
Finished  <<< sdc21x0                                   [ 0.5 seconds ]                            
Finished  <<< orocos_kdl                                [ 0.4 seconds ]                            
Starting  >>> python_orocos_kdl                                                                    
Finished  <<< mir_msgs                                  [ 0.8 seconds ]                            
Starting  >>> mir_driver                                                                           
Finished  <<< python_orocos_kdl                         [ 0.4 seconds ]                            
Starting  >>> orocos_kinematics_dynamics                                                           
Finished  <<< mir_driver                                [ 0.2 seconds ]                            
Starting  >>> mir_gazebo                                                                           
Finished  <<< orocos_kinematics_dynamics                [ 0.2 seconds ]                            
Finished  <<< mir_gazebo                                [ 0.2 seconds ]                            
Starting  >>> mir100_robot_server                                                                  
Starting  >>> mir_navigation                                                                       
Finished  <<< mir100_robot_server                       [ 0.2 seconds ]                            
Finished  <<< mir_navigation                            [ 0.1 seconds ]                            
Starting  >>> mir_robot                                                                            
Finished  <<< mir_robot                                 [ 0.1 seconds ]                            
[build] Summary: All 13 packages succeeded!                                                        
[build]   Ignored:   None.                                                                         
[build]   Warnings:  None.                                                                         
[build]   Abandoned: None.                                                                         
[build]   Failed:    None.     
```       

> source                         

```
(base) kuka-ai@kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:~/rlearning_ws$ source /opt/ros/noetic/setup.bash 
(base) kuka-ai@kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:~/rlearning_ws$ source devel/setup.bash   
(base) kuka-ai@kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:~/rlearning_ws$ printf "source /opt/ros/$ROS_DISTRO/setup.bash\nsource $ROBOGYM_WS/devel/setup.bash" >> ~/.bashrc
(base) kuka-ai@kukaai-ROG-Zephyrus-G15-GA502IU-GA502IU:~/rlearning_ws$ roslaunch mir100_robot_server sim_robot_server.launch gui:=true
  
```            

Then also I am getting this error:

[ERROR] [1623761339.206198017, 20.843000000]: GetModelState: model [mir] does not exist

matteolucchi, 2021-06-16:
I'm sorry but I am out of ideas, I tried to reinstall everything on a clean machine with Ubuntu 20.04 and ROS Noetic and the MiR environment works fine. The only other thing that I can recommend is to delete the workspace and try again to install the robot-servers part. I hope you can figure out what it is :/ 

I will close this for now as it seems to be something specific with your computer setup.

closed 2021-06-16 by matteolucchi

ranjitkathiriya, 2021-06-18:
I have solved this issue. I think that I was using a melodic branch (default branch- https://github.com/jr-robotics/mir_robot). I have switched to noetic then it was working perfectly. 

Thanks

matteolucchi, 2021-06-21:
That's great to hear!