jr-robotics/robo-gym #28: Error while retrieving state from Client

akeaveny, 2021-06-10:
Hi @matteolucchi,

I've since moved to a more powerful workstation to help me train more consistently. This workstation is at my university with a static IP address which may be causing my new issue.

I've verified the installation or the robot envs with `roslaunch` but when I run a simple script to test robo-gym I get the following:

```
Starting new Robot Server | Tentative 1
Successfully started Robot Server at localhost:40007
/home/akeaveny/anaconda3/envs/ROBOGYM/lib/python3.8/site-packages/gym/logger.py:30: UserWarning: WARN: Box bound precision lowered by casting to float32
  warnings.warn(colorize('%s: %s'%('WARN', msg % args), 'yellow'))

Episode #0 Starting!

Traceback (most recent call last):
  File "random_agent_sim.py", line 51, in <module>
    initial_observation = env.reset()
  File "/home/akeaveny/anaconda3/envs/ROBOGYM/lib/python3.8/site-packages/gym/core.py", line 289, in reset
    return self.env.reset(**kwargs)
  File "/home/akeaveny/anaconda3/envs/ROBOGYM/lib/python3.8/site-packages/gym/core.py", line 264, in reset
    observation = self.env.reset(**kwargs)
  File "/home/akeaveny/anaconda3/envs/ROBOGYM/lib/python3.8/site-packages/gym/wrappers/time_limit.py", line 25, in reset
    return self.env.reset(**kwargs)
  File "/home/akeaveny/git/robo-gym/robo_gym/envs/UWRTArm/UWRTArm.py", line 381, in reset
    rs_state = copy.deepcopy(np.nan_to_num(np.array(self.client.get_state_msg().state)))
  File "/home/akeaveny/anaconda3/envs/ROBOGYM/lib/python3.8/site-packages/robo_gym_server_modules/robot_server/client.py", line 33, in get_state_msg
    raise Exception('Error while retrieving state')
Exception: Error while retrieving state
```
Any ideas? I also tried changing localhost with the static IP address and connecting to the school's VPN. The error messgae is similar with the MiR100 or UR10 env.

Btw, I get the following output when I `roslaunch` our robot:
```
[INFO] [1623335056.455620, 10.129000]: Initializing robot_server node
[INFO] [1623335056.470300, 10.144000]:  UWRT Sim Robot Server started at 50051
```

Thank you again for you help!
Aidan

akeaveny, 2021-06-10:
Closing this as re-creating a conda env with Python 3.8.5 seemed to fix my issue - weird!

closed 2021-06-10 by akeaveny

matteolucchi, 2021-06-14:
Hi @akeaveny ! I hope you are doing well :) 

mmm yes weird, but I am glad to hear that you fixed the issue by yourself :)