jr-robotics/robo-gym #29: Port numbers that containers as both side must expose in order to communicate

wwwshwww, 2021-06-11:
Hi.

I have been using robo-gym a lot in my research on Hierarchical Deep Reinforcement Learning for mobile robots using ROS. It is a awesome tool!

I'm trying to deploy the docker containers of the robot server side and environment side on Kubernetes.
These images have been confirmed to work perfectly on docker, but I'm not sure exactly what ports each side must expose to communicate.
What is the port number that each container must explicitly expose?

matteolucchi, 2021-06-14:
Hi @wwwshwww ! It's great to hear that you have been using robo-gym! Do you have any publications/public work ?

I've never used Kubernetes myself but I guess this doesn't matter for the question you asked. 
We use the Docker containers of [robo-gym](https://github.com/jr-robotics/robo-gym) and [robo-gym-robot-servers](https://github.com/jr-robotics/robo-gym-robot-servers) in our CI and to test the environments we use docker-compose and start the the 2 containers with a shared network so we don't need to explicitly define the ports to expose. 

Nevertheless, it is possible to narrow down the port range that the Server Manager and the robot-servers use.

The Server Manager uses the first free port in the range [50100 to 50200](https://github.com/jr-robotics/robo-gym-server-modules/blob/5a30009482eb82375f681720e4f1146174b34bc6/robo_gym_server_modules/server_manager/server.py#L162).

The Robot Servers are currently started on a random free port, but you could easily restrict the range of ports used [here](https://github.com/jr-robotics/robo-gym-server-modules/blob/5a30009482eb82375f681720e4f1146174b34bc6/robo_gym_server_modules/server_manager/server.py#L69).

I hope this helps :) Let me know if you have further questions

wwwshwww, 2021-06-25:
Hi @matteolucchi. Thank you for replying!
I have just recently submitted a short paper on intelligent architecture of robots, citing robo-gym, to FIT2021 in Japan.

As it turns out, the problem has been solved.
I certainly didn't have to think about exposing explicit ports in terms of Docker or Kubernetes specifications.

The problem with the robot server not starting was due to the fact that the robo-gym workspace was not overlaid. The problem with the robot server not starting was due to the fact that the robo-gym workspace was not overlaid. Reason was that there was no `source /robogym_ws/devel/setup.bash` command in the manifest file of the Kubernetes resource that runs the robot server container.

In the end, I confirmed that we can use robo-gym without any problem on Kubernetes by using a manifest for robot server defined as the following example.

```YAML
apiVersion: v1
kind: Pod
metadata:
  name: robo-gym-robot-servers
spec:
  containers:
  - name: robo-gym-robot-servers
    image: robot-servers
    command: ["bash"]
    args: ["-c", "source /robogym_ws/devel/setup.bash && start-server-manager && tail -f /dev/null"]
```

Thanks!

closed 2021-06-25 by wwwshwww

matteolucchi, 2021-06-25:
Hi @wwwshwww, 

> I have just recently submitted a short paper on intelligent architecture of robots, citing robo-gym, to FIT2021 in Japan.

Thank you for citing our work and good luck with your submission!

> In the end, I confirmed that we can use robo-gym without any problem on Kubernetes by using a manifest for robot server defined as the following example.

That's great to know, thank you for reporting back, it's useful for us to know this!

Cheers,

Matteo