jr-robotics/robo-gym #32: How to start simulation

Miftahur92, 2021-06-27:
Hi, this might be a very silly question.
I am running Ubuntu 18, ROS melodic and Python 3.6.9.
I installed robo-gym and robo-gym-server (both master branch) and no error during installation.
In terminal 1, i tried >> _start-server-manager_ but nothing happens. then I tried _roslaunch ur_robot_server ur_robot_server.launch ur_model:=ur10  gui:=true_ and then only RVIZ comes up with a UR10 arm.
In terminal 2, i tried _import gym, robo_gym_ but I get an error **import-im6.q16: not authorized `robo-gym' @ error/constitute.c/WriteImage/1037."** 
If I try to run _env = gym.make('EmptyEnvironmentURSim-v0', ur_model='ur5', ip='127.0.0.1')_, I get syntax error, "bash: syntax error near unexpected token `('".
please, help me to run the simulation for a UR10.

matteolucchi, 2021-06-30:
HI @Miftahur92 and thank you for using robo-gym!

> i tried >> start-server-manager but nothing happens.
Please have a look at the [Testing section](https://github.com/jr-robotics/robo-gym#testing) of the README and report back on the results you get. 

> In terminal 2, i tried import gym, robo_gym but I get an error **import-im6.q16: not authorized robo-gym' @ error/constitute.c/WriteImage/1037."** If I try to run _env = [...]

Are you sure you ran this in Python3?

Cheers, 
Matteo

Miftahur92, 2021-07-05:
Hi @matteolucchi , thanks for the reply.
Outcome of  >> start-server-manager && attach-to-server-manager is as expected.
But, there was some error with pytest. please, check attachment 1.
[image]

I have created a pyenv environment and checked the python version after activating it. Python version is 3.6.9. I found my mistake, I was directly running them in the terminal instead of starting python first. 
I tried to run the random_agent_sim.py. Rviz and Gazebo come up but no robot. Please check attachment 2.
What should I do next to start the training?
[image]

matteolucchi, 2021-07-13:
Hi @Miftahur92 !

> But, there was some error with pytest. please, check attachment 1.

I think that pytest was run from the wrong directory, it is not specified in the docs. Please go to run `pytest` from the `robo-gym` folder (while the correct python environment for robo-gym is activated).

> I have created a pyenv environment and checked the python version after activating it. Python version is 3.6.9. I found my mistake, I was directly running them in the terminal instead of starting python first.
I tried to run the random_agent_sim.py. Rviz and Gazebo come up but no robot. Please check attachment 2.
What should I do next to start the training?

That is very weird, let's try to just launch the Mir100 simulation and see why the robot model is not loaded. Please try to run:
```
roslaunch mir_gazebo mir_empty_world.launch
```
and let me know what's the output. 

Cheers, 
Matteo

Miftahur92, 2021-07-14:
@matteolucchi **Thnaks for reply. Part of result from Pytest:**
_tests/robo-gym/envs/mir100/test_mir100_base_env.py E                                                                                                                                           [  0%]
tests/robo-gym/envs/mir100/test_mir100_obstacle_avoidance.py E                                                                                                                                 [  0%]
tests/robo-gym/envs/ur/test_ur_avoidance_basic.py EsEEEsEEEsEEEsEEEsEEEsEEEsEEFFFFFFF                                                                                                          [ 17%]
tests/robo-gym/envs/ur/test_ur_avoidance_iros_fixed_spline.py ssssssEsEssssssssssssF^[[A^[[B^[[B^C

=============================================================================================== ERRORS ===============================================================================================
_______________________________________________________________________________ ERROR at setup of test_initialization ________________________________________________________________________________

request = <SubRequest 'env' for <Function test_initialization>>

    @pytest.fixture(scope='module')
    def env(request):
>       env = gym.make('NoObstacleNavigationMir100Sim-v0', ip='robot-servers')

tests/robo-gym/envs/mir100/test_mir100_base_env.py:7: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../../../.pyenv/versions/3.6.9/envs/robo-gym/lib/python3.6/site-packages/gym/envs/registration.py:156: in make
    return registry.make(id, **kwargs)
../../../.pyenv/versions/3.6.9/envs/robo-gym/lib/python3.6/site-packages/gym/envs/registration.py:101: in make
    env = spec.make(**kwargs)
../../../.pyenv/versions/3.6.9/envs/robo-gym/lib/python3.6/site-packages/gym/envs/registration.py:73: in make
    env = cls(**_kwargs)
robo_gym/envs/mir100/mir100.py:411: in __init__
    Simulation.__init__(self, self.cmd, ip, lower_bound_port, upper_bound_port, gui, **kwargs)
robo_gym/envs/simulation_wrapper.py:25: in __init__
    self.sm_client = sm_client.Client(ip)
../../../.pyenv/versions/3.6.9/envs/robo-gym/lib/python3.6/site-packages/robo_gym_server_modules/server_manager/client.py:12: in __init__
    assert self._connect_to_rl_srv_mng(ip,lower_bound_port,upper_bound_port)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <robo_gym_server_modules.server_manager.client.Client object at 0x7f3b34c3e668>, ip = 'robot-servers', lower_bound_port = 50100, upper_bound_port = 50200

    def _connect_to_rl_srv_mng(self,ip,lower_bound_port,upper_bound_port):
    
        for port in range(lower_bound_port,upper_bound_port):
            try:
                self.channel = grpc.insecure_channel('{}:{}'.format(ip,str(port)))
                self.stub = server_manager_pb2_grpc.ServerManagerStub(self.channel)
                self.ip = ip
                self._verify_connection()
                return True
            except RpcError as rpc_error:
                pass
>       raise RuntimeError('Failed to connect to Server Manager')
E       RuntimeError: Failed to connect to Server Manager

../../../.pyenv/versions/3.6.9/envs/robo-gym/lib/python3.6/site-packages/robo_gym_server_modules/server_manager/client.py:83: RuntimeError
_______________________________________________________________________________ ERROR at setup of test_initialization ________________________________________________________________________________

request = <SubRequest 'env' for <Function test_initialization>>

    @pytest.fixture(scope='module')
    def env(request):
>       env = gym.make('ObstacleAvoidanceMir100Sim-v0', ip='robot-servers')

tests/robo-gym/envs/mir100/test_mir100_obstacle_avoidance.py:7: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../../../.pyenv/versions/3.6.9/envs/robo-gym/lib/python3.6/site-packages/gym/envs/registration.py:156: in make
    return registry.make(id, **kwargs)
../../../.pyenv/versions/3.6.9/envs/robo-gym/lib/python3.6/site-packages/gym/envs/registration.py:101: in make
    env = spec.make(**kwargs)
../../../.pyenv/versions/3.6.9/envs/robo-gym/lib/python3.6/site-packages/gym/envs/registration.py:73: in make
    env = cls(**_kwargs)
robo_gym/envs/mir100/mir100.py:630: in __init__
    Simulation.__init__(self, self.cmd, ip, lower_bound_port, upper_bound_port, gui, **kwargs)
robo_gym/envs/simulation_wrapper.py:25: in __init__
    self.sm_client = sm_client.Client(ip)
../../../.pyenv/versions/3.6.9/envs/robo-gym/lib/python3.6/site-packages/robo_gym_server_modules/server_manager/client.py:12: in __init__
    assert self._connect_to_rl_srv_mng(ip,lower_bound_port,upper_bound_port)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

[... cut ...]
E       RuntimeError: Failed to connect to Server Manager

../../../.pyenv/versions/3.6.9/envs/robo-gym/lib/python3.6/site-packages/robo_gym_server_modules/server_manager/client.py:83: RuntimeError_

I treid to launch mir_empty_world and it launched without any error.

But, I have to change the platform from melodic to kinetic. We have a mobile manipulator which is running in kinetic. So, I am planning to install robo-gym in kinetic. 
If you want, you can close this issue. I will create new issue if I issue implementing robo-gym in kinetic.

matteolucchi, 2021-07-15:
Ok, just remind this:

> WARNING for ROS Kinetic users
The compatibility of the Universal Robots environments with ROS Kinetic has been maintained until version v0.1.8 included. After this version we integrated a refactored version of the universal_robot repository which is not compatible with ROS Kinetic. To use the UR environments on ROS kinetic you need to use v0.1.8 of the robo-gym package and v.0.1.8 of the robo-gym-robot-servers. See #16 for more details.

As noted in https://github.com/jr-robotics/robo-gym-robot-servers#readme

Cheers, 
Matteo

closed 2021-07-15 by matteolucchi