jr-robotics/robo-gym #38: Running in ROS kinetic

Miftahur92, 2021-07-15:
Dear @matteolucchi 
I tried to install robo-gym in ROS kinetic.
1. First I used the following commands to install robo-gym-server (v0.1.8):
_mkdir -p ~/robogym_ws/src && cd ~/robogym_ws/src
git clone https://github.com/jr-robotics/robo-gym-robot-servers.git
cd ~/robogym_ws/src/robo-gym-robot-servers
git checkout v0.1.8
./kinetic-install.sh_
This is the summary of installation.
[image]

2. Then I installed robo-gym-server-modules using following command: 
_pip install robo-gym-server-modules_

3. At last I created a pyenv virtual environment with python 3.5.9 and installed robo-gym using following command:
_git clone https://github.com/jr-robotics/robo-gym.git
cd robo-gym
git checkout v0.1.8
pip install ._

Problems:
1. I ran start-server-manager and then attach-to-server-manager, I get no session message.
[image]

2. I manually launched mir100 server using roslaunch and there was no problem

3.  Then tried to create env enabling python 3.5.2 but got error
[image]

4. From one of  issue and answers, I tried to run client.py from server_manager and got error
[image]

How can I solve this? Thanks in advance.

matteolucchi, 2021-08-19:
Hello @Miftahur92 ! Sorry for the late reply but somehow we missed this issue. 

Please try to uninstall `robo-gym-server-modules` and install the version 0.1.2 with:

```
pip install robo-gym-server-modules==0.1.2
```

I hope this will fix it. 

Cheers, 
Matteo

Miftahur92, 2021-08-23:
Thanks @matteolucchi . It worked.

closed 2021-08-23 by Miftahur92

matteolucchi, 2021-08-23:
Great, I am glad to hear that!