jr-robotics/robo-gym #42: Cannot run random_agent_sim

psFournier, 2021-08-16:
Hello again!

roslaunch mir100_robot_server sim_robot_server.launch gui:=true works fine with my install, but for some reason, python3 docs/examples/random_agent_sim.py cannot start a new Robot Server.

Precisely, checking the roslaunch log with tmux -L ServerManager a -t 44313 with 44313 being the port found by the add_rl_server, I get:

process[master]: started with pid [2066080]                                       
ROS_MASTER_URI=http://localhost:38063                                                                
                                                                                                     
setting /run_id to a38e3900-fea3-11eb-9901-f1490f2031de                           
process[rosout-1]: started with pid [2066090]              
started core service [/rosout]                                                                       
process[gazebo-2]: started with pid [2066098]                 
/opt/ros/noetic/lib/gazebo_ros/gzserver: 1: catkin_find: not found
/opt/ros/noetic/lib/gazebo_ros/gzserver: 1: catkin_find: not found                                   
process[spawn_urdf-3]: started with pid [2066106]             
process[controller_spawner-4]: started with pid [2066120]         
process[ekf_localization_node-5]: started with pid [2066136]                                         
process[joint_state_publisher-6]: started with pid [2066143]       
process[robot_state_publisher-7]: started with pid [2066145]     
process[b_scan_relay-8]: started with pid [2066150]               
process[f_scan_relay-9]: started with pid [2066152]                
process[b_rep117_laser_filter-10]: started with pid [2066161]    
process[f_rep117_laser_filter-11]: started with pid [2066164] 
process[fake_localization-12]: started with pid [2066169]                
process[robot_pose_publisher-13]: started with pid [2066170]       
process[cmd_vel_command_handler-14]: started with pid [2066173]
process[rviz_qdtis056z_2066067_7750190536441384180-15]: started with pid [2066178]
process[robot_server-16]: started with pid [2066180]               
[ INFO] [1629126435.751530965]: rviz version 1.14.8            
[ INFO] [1629126435.751569966]: compiled against Qt version 5.12.8                
[ INFO] [1629126435.751580032]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1629126435.768650157]: Forcing OpenGl version 0.  
**[INFO] [1629126435.783113, 0.000000]: Controller Spawner: Waiting for service controller_manager/load
_controller                                                                                          
Error. Invalid arguments                                   
[gazebo-2] process has died [pid 2066098, exit code 255, cmd /opt/ros/noetic/lib/gazebo_ros/gzserver 
-e ode /d/pfournie/Documents/paradis/robo-gym-ws/src/robo-gym-robot-servers/mir100_robot_server/world
s/empty_world_target.world joint_states:=mir/joint_states mobile_base_controller/cmd_vel:=cmd_vel mob
ile_base_controller/odom:=odom_comb __name:=gazebo __log:=/d/pfournie/.ros/log/a38e3900-fea3-11eb-990
1-f1490f2031de/gazebo-2.log].                                                                        
log file: /d/pfournie/.ros/log/a38e3900-fea3-11eb-9901-f1490f2031de/gazebo-2*.log**                    
[INFO] [1629126436.102571, 0.000000]: Waiting 10s before starting initialization of robot_server     
[ INFO] [1629126436.346782946]: Stereo is NOT SUPPORTED                                              
[ INFO] [1629126436.346824328]: OpenGL device: NVIDIA GeForce RTX 2080 SUPER/PCIe/SSE2
[ INFO] [1629126436.346842031]: OpenGl version: 4,6 (GLSL 4,6).                                 
[WARN] [1629126466.009456, 0.000000]: Controller Spawner couldn't find the expected controller_manage
r ROS interface.                                                                                     
[controller_spawner-4] process has finished cleanly            
log file: /d/pfournie/.ros/log/a38e3900-fea3-11eb-9901-f1490f2031de/controller_spawner-4*.log

For some other reason, I can't find any file with the log from the gazebo-2 process...

Once again, this problem pops up on one of my two machines only if it can help...

Thank you in advance !

psFournier, 2021-08-16:
If it can be of any help, the failing command launches rviz with this result:
[image]

matteolucchi, 2021-08-23:
Hello @psFournier !

I am a bit out ideas here, especially because you mentioned that running `roslaunch mir100_robot_server sim_robot_server.launch gui:=true` works fine. 
Is it possible that you are running the 2 test cases in 2 different python virtual environments? 

When running `roslaunch mir100_robot_server sim_robot_server.launch gui:=true` manually, is the robot visualised correctly?

psFournier, 2021-08-23:
Hello,
I just found the solution: once again, the culprit was something specific to the computer on which the issue appeared (some line in my bashrc I don't remember having put myself was messing with the PATH env variable in tmux sessions)... I am slowly discovering ros, tmux and grpc , i can say it's not always easy to pinpoint the nature and origin of bugs :D I sincerely hope I did not make you search too long for this non-reproducible issue...
Thank you for your time !

matteolucchi, 2021-08-23:
> some line in my bashrc I don't remember having put myself was messing with the PATH env variable in tmux sessions  

Yes that makes sense :)

>  i can say it's not always easy to pinpoint the nature and origin of bugs

Yes unfortunately that is the case, but we are slowly trying to improve that :)

> I sincerely hope I did not make you search too long for this non-reproducible issue...

No worries, I am glad that the issue is fixed now :)

closed 2021-08-23 by matteolucchi