jr-robotics/robo-gym #44: Error occurred while calling the reset function. Restarting Robot server

Papul-Ghosh, 2021-09-02:
Hi, I am trying the mir100 example for UR10 robot.

This was my code:

import gym
import robo_gym
from robo_gym.wrappers.exception_handling import ExceptionHandling

# initialize environment
env = gym.make('EmptyEnvironmentURSim-v0', ur_model='ur10', ip='127.0.0.1', gui=True)
env = ExceptionHandling(env)

num_episodes = 10

for episode in range(num_episodes):
    done = False
    env.reset()
    while not done:
        # random step in the environment
        state, reward, done, info = env.step(env.action_space.sample())

Although I ran each command individually. Each command till "num_episodes = 10" was okay. Gazebo environment is also showing perfect.

But when I execute the last for loop snippet, this error shows up:

"
Error occurred while calling the reset function. Restarting Robot server ...
Killing Robot Server at 127.0.0.1:44009 | Tentative 1 of 10
Successfully killed Robot Server at 127.0.0.1:44009
Starting new Robot Server | Tentative 1 of 10
Successfully started Robot Server at 127.0.0.1:42945
Killing Robot Server at 127.0.0.1:44009 | Tentative 1 of 10
Successfully killed Robot Server at 127.0.0.1:44009
Starting new Robot Server | Tentative 2 of 10
.
.
.
"
and it continues
the same happens for all UR environments (mir works fine). Please help me solving the issue for UR.

matteolucchi, 2021-09-06:
Hi @Papul-Ghosh ! 

Which Ubuntu version are you using? And which ROS distribution?

Cheers,
Matteo

Papul-Ghosh, 2021-09-06:
Hi @matteolucchi,

It's Ubuntu 20.04 with ros noetic,

Regards,
Papul

matteolucchi, 2021-09-06:
I tried to reproduce your error but it works fine on our PC. 

After you started the environment please open a new terminal window and run:
`tmux -L ServerManager`
this will attach a tmux session to the tmux server running the RobotServer in the background. 
Now to navigate to the session where the RobotServer is running use `Ctrl + B    )`
You should see something like:
```
[ INFO] [1630915846.414570295]: Starting gazebo_ros_control plugin in namespace: /
[ INFO] [1630915846.415445293]: gazebo_ros_control plugin is waiting for model URDF in parameter [robot_description] on the ROS param server.
[ INFO] [1630915846.537557683]: Loaded gazebo_ros_control.
[INFO] [1630915846.644456, 0.000000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1630915846.646374, 0.000000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1630915846.648065, 0.000000]: Loading controller: joint_group_eff_controller
[INFO] [1630915846.659470, 0.000000]: Controller Spawner: Loaded controllers: joint_group_eff_controller
[INFO] [1630915846.697643, 0.000000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1630915846.699547, 0.000000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1630915846.701191, 0.000000]: Loading controller: joint_state_controller
[INFO] [1630915846.707146, 0.000000]: Loading controller: eff_joint_traj_controller
[INFO] [1630915846.806982, 0.000000]: Controller Spawner: Loaded controllers: joint_state_controller, eff_joint_traj_controller
[INFO] [1630915847.390970, 0.000000]: Calling service /gazebo/set_model_configuration
[INFO] [1630915847.400466, 0.000000]: Set model configuration status: SetModelConfiguration: success
[INFO] [1630915847.404106, 0.000000]: Unpausing physics
[INFO] [1630915847.418314, 0.001000]: Started controllers: joint_state_controller, eff_joint_traj_controller
[spawn_gazebo_model-5] process has finished cleanly
log file: /home/robotics/.ros/log/ee5a0e68-0ee9-11ec-b989-45957c9a2d9f/spawn_gazebo_model-5*.log
[INFO] [1630915852.426883, 5.001000]: Starting UR Robot Server...
[INFO] [1630915852.487701, 5.062000]: ur10 Sim Robot Server started at 46347
```

To be able to scroll through the output use:
`Ctrl+B   [`

Please copy the output and report it here so that I can see what is going on. 

If you have some troubles with the tmux commands you can have a look here https://tmuxcheatsheet.com/

Papul-Ghosh, 2021-09-07:
Thanks for the assistance.

So first I executed:
```
start-server-manager
```
Then this code snippet:
```
import gym
import robo_gym
from robo_gym.wrappers.exception_handling import ExceptionHandling
env = gym.make('EmptyEnvironmentURSim-v0', ur_model='ur10', ip='127.0.0.1', gui=True)
```
Then on a new terminal window:
```
tmux -L ServerManager
```

The following was the output of the tmux command in my system:

```
[ INFO] [1631031522.422928563]: Loading gazebo_ros_control plugin
[ INFO] [1631031522.423072558]: Starting gazebo_ros_control plugin in namespace: /
[ INFO] [1631031522.424180990]: gazebo_ros_control plugin is waiting for model URDF in parameter [robot_description] on the ROS param server.
[ INFO] [1631031522.547316606]: Loaded gazebo_ros_control.
[INFO] [1631031522.552407, 0.000000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1631031522.555326, 0.000000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1631031522.558414, 0.000000]: Loading controller: joint_group_eff_controller
[ERROR] [1631031522.562309906]: Could not load controller 'joint_group_eff_controller' because controller type 'effort_controllers/JointGroupEffortController' does not exist.
[ERROR] [1631031522.562365445]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[INFO] [1631031522.810631, 0.000000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1631031522.813458, 0.000000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1631031522.816621, 0.000000]: Loading controller: joint_state_controller
[INFO] [1631031522.829738, 0.000000]: Loading controller: eff_joint_traj_controller
[ERROR] [1631031522.834060033]: Could not load controller 'eff_joint_traj_controller' because controller type 'effort_controllers/JointTrajectoryController' does not exist.
[ERROR] [1631031522.834110539]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[INFO] [1631031523.394777, 0.000000]: Calling service /gazebo/set_model_configuration
[INFO] [1631031523.409068, 0.000000]: Set model configuration status: SetModelConfiguration: success
[INFO] [1631031523.413301, 0.000000]: Unpausing physics
[ERROR] [1631031523.564402, 0.049000]: Failed to load joint_group_eff_controller
[INFO] [1631031523.570625, 0.049000]: Controller Spawner: Loaded controllers: 
[ERROR] [1631031523.836566, 0.161000]: Failed to load eff_joint_traj_controller
[INFO] [1631031523.843338, 0.161000]: Controller Spawner: Loaded controllers: joint_state_controller
[INFO] [1631031523.874964, 0.164000]: Started controllers: joint_state_controller
[spawn_gazebo_model-5] process has finished cleanly
log file: /home/papul/.ros/log/424c03b0-0ff7-11ec-9cfc-314dfeb981c4/spawn_gazebo_model-5*.log
[INFO] [1631031541.772535, 5.001000]: Starting UR Robot Server...
[INFO] [1631031542.170568, 5.084000]: ur10 Sim Robot Server started at 39137
```

So you can see there are some errors loading controllers 'joint_group_eff_controller' and 'eff_joint_traj_controller'.
Can you guess where is the issue?

Thanks

matteolucchi, 2021-09-08:
Thank you, I guess some of the packages did not get installed properly during the installation process. 

Please try:
```
sudo apt-get update
sudo apt install ros-noetic-effort-controllers ros-noetic-joint-trajectory-controller
```

Let me know if this fixes the issue.

Cheers,
Matteo

Papul-Ghosh, 2021-09-08:
Thanks a lot @matteolucchi. it worked 😍. Now I understand the issue.
Just had a last query, only the UR is is visible in the world, not the object. is it normal?
Once again thanks 🍻

matteolucchi, 2021-09-08:
That's great to hear! 

> Just had a last query, only the UR is is visible in the world, not the object. is it normal?

If you are using the `EmptyEnvironmentURSim-v0` environment, as you can see [here](https://github.com/jr-robotics/robo-gym/blob/master/docs/environments.md#empty-environment) this is normal.

You are welcome :)

matteolucchi, 2021-09-08:
If this is solved, please close the issue :)

Papul-Ghosh, 2021-09-08:
Thanks for all the assistance and clarification.

closed 2021-09-08 by Papul-Ghosh

ecada, 2021-11-01:
Hello Matteo, I am still having the same issue with Ubuntu 20.04.3 and ros noetic. The tmux does not show anything although the simulation runs briefly. I also tried the following with no luck: 

sudo apt-get update
sudo apt install ros-noetic-effort-controllers ros-noetic-joint-trajectory-controller

The same occurs in UR10 and NoObstacleNavigationMir100Sim-v0

The UR10 environment gives the following error when I run the code 1 by 1 and the same error as Papul-Ghosh

>>> env.reset()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/**/.local/lib/python3.8/site-packages/gym/wrappers/order_enforcing.py", line 16, in reset
    return self.env.reset(**kwargs)
  File "/home/**/robo-gym/robo_gym/envs/ur/ur_base_env.py", line 118, in reset
    raise InvalidStateError()
robo_gym.utils.exceptions.InvalidStateError: The environment state received is not contained in the observation space.

ecada, 2021-11-01:
Below is the error from NoObstacleNavigationMir100Sim-v0 environment:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/**/.local/lib/python3.8/site-packages/gym/wrappers/order_enforcing.py", line 16, in reset
    return self.env.reset(**kwargs)
  File "/home/**/robo-gym/robo_gym/envs/mir100/mir100.py", line 114, in reset
    raise InvalidStateError()
robo_gym.utils.exceptions.InvalidStateError: The environment state received is not contained in the observation space.

matteolucchi, 2021-11-02:
Hello @ecada,

Have you tried to do what was done here https://github.com/jr-robotics/robo-gym/issues/44#issuecomment-914456650https://github.com/jr-robotics/robo-gym/issues/44#issuecomment-914456650 ?

> The tmux does not show anything although the simulation runs briefly.

This also sounds like something went wrong during the installation process, can you please try to install again the robo-gym-robot-servers ?

Cheers, 
Matteo