jr-robotics/robo-gym #45: GAZEBO_MASTER_URI

posshe, 2021-09-14:
Dear @matteolucchi :
    Hi, I have been using robo-gym in my work about Deep Reinforcement Learning. It is really a awesome tool!
    I have a question about "server_manager/server.py", the GAZEBO_MASTER_URI is all set on the localhost . Is it means all gzservers works on the localhost of the server_manager? If so, does it mean the amount of gazebo calculation mainly exists in the localhost of the server_manager?

`        gazebo_port = find_free_port()
        server_pane.send_keys(cmd= 'export GAZEBO_MASTER_URI=http://localhost:{}'.format(repr(gazebo_port)), suppress_history = False)`

    Looking forward to your reply.
    Thank you.

posshe, 2021-09-14:
The question is based on the assumption : There are several computers using the gazebo to generator the RL data.
Thank you!

matteolucchi, 2021-09-14:
Hi @posshe ! Thank you for your kind words, I am glad to hear that you are using robo-gym! 

Yes, you are correct. Just bear in mind that you can have multiple server managers running on different computers. Just to give you an example, you could have 2 different pcs on which you start the server manager, these will be the ones where the gazebo simulations will be performed, and you could have a third machine on which you run the learning algorithm and start the robo-gym environments that connect to the 2 different computers running the server managers. 
I hope this helps you. 

Cheers, 
Matteo

posshe, 2021-09-15:
Dear @matteolucchi :
Thank you for the nice reply.  I got a wrong understanding that the amount limit of the server managers  is only one. At this time I understand. Thank you for the excellent work again and you can close the issue.
Thank you,
posshe

closed 2021-09-15 by posshe