jr-robotics/robo-gym #46: How to use it correctly?

KeisukeHamamoto, 2021-10-04:
Hi,

I have a question for you from Japan.
First of all, please understand that my English is not good.

I was looking for a reinforcement learning environment.
I think robo-gym is the most suitable for my tasks so I want to use it.
But I could not use it well so I want to know how to use it from start to end.

In 1st terminal
roslaunch ur_robot_server ur_robot_server.launch ur_model:=ur10 gazebo_gui:=true gui:=true
In 2nd terminal
start-server-manager && attach-to-server-manager
In 3rd terminal
python3 main.py

I did this but I thought it did not start learning.

matteolucchi, 2021-10-05:
Hi @KeisukeHamamoto and thank you for your interest in robo-gym!

There is no need to start `roslaunch ur_robot_server ur_robot_server.launch ur_model:=ur10 gazebo_gui:=true gui:=true`

It is enough to start the following:
terminal 1:
`start-server-manager && attach-to-server-manager`
terminal 2:
`python3 main.py`

assuming that main.py is your script. 

I hope this helps.

Cheers, 
Matteo

KeisukeHamamoto, 2021-10-05:
Thank you for the quick reply!
 
I'll try it.

closed 2021-10-05 by KeisukeHamamoto