jr-robotics/robo-gym #49: Error for running example

pouyan-asg, 2021-11-05:
Hello
after a lot of searches I found your repository that can work with noetic (for RL).
I just wanted to test a simple example as you put inside your README section. I just get this error:

Error occurred while calling the reset function. Restarting Robot server ...
Killing Robot Server at 127.0.0.1:48025 | Tentative 1 of 10
Successfully killed Robot Server at 127.0.0.1:48025
Starting new Robot Server | Tentative 1 of 10
Successfully started Robot Server at 127.0.0.1:36809
Error occurred while calling the reset function. Restarting Robot server ...
Killing Robot Server at 127.0.0.1:36809 | Tentative 1 of 10

it repeats.
I am using VS code and my system meets all requirements that you mentioned.
How I can tackle this issue?

thank you in advance

Shtaiven, 2021-11-11:
I am getting the exact same error. The gym environment is created on a machine running Pop!_OS 21.04 (based on Ubuntu 21.04) with python version 3.9.5

The server is installed from the master branch of `robo-gym-robot-servers` on a clean install of Ubuntu 20.04 with ROS Noetic

matteolucchi, 2021-12-23:
Hi @pouyan-asg and @Shtaiven ! 

Thank you very much for using robo-gym and sorry for the late reply! I only had time now to look into the issue and some changes in the gym package caused the environments to break. 
It should be fixed now, if you are stil interested please  do a fresh installation and try it out and feel free to reopen the issue in case it is still not working. 

Cheers, 

Matteo

closed 2021-12-23 by matteolucchi

Alvin-shao, 2023-07-27:
> 嗨和！
> 
> 非常感谢您使用机器人健身房，很抱歉回复晚了！我现在只有时间研究这个问题，健身房套餐中的一些变化导致环境破裂。现在应该修复它，如果您仍然感兴趣，请进行全新安装并尝试一下，如果它仍然无法正常工作，请随时重新打开问题。
> 
> 干杯
> 
> 马迪奥

I have been experiencing this issue and cannot solve it. Other installations have not had any problems, except for the reset function call error.

Alvin-shao, 2023-07-28:
> Hello after a lot of searches I found your repository that can work with noetic (for RL). I just wanted to test a simple example as you put inside your README section. I just get this error:
>
> Error [...]

May I ask if you have solved the problem? My problem is the same as yours. How did you solve it?