jr-robotics/robo-gym #50: How do you debug the ros_bridge in general ?

psFournier, 2021-11-16:
Hello,
Not an issue per se but a simple question : what solution would you or did you find more practical to debug a custom ros_bridge for another robot, given that its code is run in the grpc servicer created by the robot_server node ?
Thank you !

matteolucchi, 2021-11-17:
Hi @psFournier , 
it depends on what part you want to debug. The Robot Server is started in a separate [tmux](https://github.com/tmux/tmux) session,  after you started the environment please open a new terminal window and run:
`tmux -L ServerManager`
this will attach a tmux session to the tmux server running the Robot Server in the background.
Now to navigate to the session where the RobotServer is running use `Ctrl + B )`
You should see something like:

```
[ INFO] [1630915846.414570295]: Starting gazebo_ros_control plugin in namespace: /
[ INFO] [1630915846.415445293]: gazebo_ros_control plugin is waiting for model URDF in parameter [robot_description] on the ROS param server.
[ INFO] [1630915846.537557683]: Loaded gazebo_ros_control.
[INFO] [1630915846.644456, 0.000000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1630915846.646374, 0.000000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1630915846.648065, 0.000000]: Loading controller: joint_group_eff_controller
[INFO] [1630915846.659470, 0.000000]: Controller Spawner: Loaded controllers: joint_group_eff_controller
[INFO] [1630915846.697643, 0.000000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1630915846.699547, 0.000000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1630915846.701191, 0.000000]: Loading controller: joint_state_controller
[INFO] [1630915846.707146, 0.000000]: Loading controller: eff_joint_traj_controller
[INFO] [1630915846.806982, 0.000000]: Controller Spawner: Loaded controllers: joint_state_controller, eff_joint_traj_controller
[INFO] [1630915847.390970, 0.000000]: Calling service /gazebo/set_model_configuration
[INFO] [1630915847.400466, 0.000000]: Set model configuration status: SetModelConfiguration: success
[INFO] [1630915847.404106, 0.000000]: Unpausing physics
[INFO] [1630915847.418314, 0.001000]: Started controllers: joint_state_controller, eff_joint_traj_controller
[spawn_gazebo_model-5] process has finished cleanly
log file: /home/robotics/.ros/log/ee5a0e68-0ee9-11ec-b989-45957c9a2d9f/spawn_gazebo_model-5*.log
[INFO] [1630915852.426883, 5.001000]: Starting UR Robot Server...
[INFO] [1630915852.487701, 5.062000]: ur10 Sim Robot Server started at 46347
```
To be able to scroll through the output use:
`Ctrl+B [`

If you have some troubles with the tmux commands you can have a look here https://tmuxcheatsheet.com/

psFournier, 2021-11-17:
Sorry, I should have been more specific : by debugging I meant using something like gdb and more generally being able to set breakpoints !
I haven't investigated any solution yet, but to me the ros_bridge code being run exclusively in grpc methods  makes it difficult to debug with standard methods.
Initially this question appeared when I was trying to understand why some callbacks like callbackOdometry do not check for get_state_event.isSet() in the mir100 ros bridge.

matteolucchi, 2021-12-23:
Hi @psFournier, 

sorry for the late reply, I am afraid I cannot help you here, I have never used anything like gdb or debug tool so far, I agree with you that it is not an optimal way to debug but we haven't had time to look into the issue so far. If you have any suggestions or want to contribute something to the repo, everything is more than welcome. 

Cheers, 

Matteo

closed 2021-12-23 by matteolucchi