jr-robotics/robo-gym #51: [BUG] Example of `NoObstacleNavigationMir100Sim-v0` isn't working

Rowing0914, 2021-12-01:
Hi,

Thank you for your effort to make the community better by providing this support for DRL for robotics in industry!
I've found some mistake in the logic such that the provided example isn't working.

- Here you allows self.client to be None: [Code](https://github.com/jr-robotics/robo-gym/blob/master/robo_gym/envs/mir100/mir100.py#L53-L57)
- But, for instance, [here](https://github.com/jr-robotics/robo-gym/blob/master/robo_gym/envs/mir100/mir100.py#L99) or [there](https://github.com/jr-robotics/robo-gym/blob/master/robo_gym/envs/mir100/mir100.py#L103), self.client is called such that this ends up causing the error...

Anyway, the primary cause is this vairable being set to None.
So could you fix this bug?

- https://github.com/jr-robotics/robo-gym/blob/master/robo_gym/envs/simulation_wrapper.py#L17

Rowing0914, 2021-12-01:
It turned out that this seems a bit critical bug....
Without self.client, there seems to be no way to execute an action in step method, right?

Rowing0914, 2021-12-01:
My test code.

```python
import os
import gym
import robo_gym
from robo_gym.wrappers.exception_handling import ExceptionHandling

target_machine_ip = '127.0.0.1'  # or other machine 'xxx.xxx.xxx.xxx'

# initialize environment
env = gym.make('NoObstacleNavigationMir100Sim-v0', ip=target_machine_ip, gui=False)
env = ExceptionHandling(env)

num_episodes = 1

for episode in range(num_episodes):
    done = False
    env.reset()
    for i in range(10):
        # random step in the environment
        action = env.action_space.sample()
        state, reward, done, info = env.step(action)
        print(i, state, action, reward, done)
```

Rowing0914, 2021-12-01:
just in case, this is the result of print in my test code.

```shell
0 [ 1.34030803e+00 -1.20318812e+00 -1.74082124e-05 -7.84278018e-05] [ 0.9253407 -0.312263 ] -0.2869701039791107 False
1 [ 1.3365632  -1.19848513  0.12905604 -0.15167511] [ 0.06061185 -0.8371541 ] 0.1439434077801735 False
2 [ 1.33320738 -1.17025311  0.07520421 -0.35140991] [-0.07234666 -0.2972456 ] 0.13716978840021568 False
3 [ 1.33320738 -1.17025311 -0.03696184 -0.31456628] [-0.01025574  0.02454386] -0.0038130378909409044 False
4 [ 1.33458648 -1.13710905 -0.03696184 -0.31456628] [ 0.8735261  -0.25257793] -0.33859045229822354 False
5 [ 1.33486152 -1.12828156 -0.00371993 -0.08330178] [0.70487934 0.7882005 ] -0.2488615449624916 False
6 [ 1.33064721 -1.12499685  0.0982857  -0.10691188] [-0.52650326  0.33365813] 0.042754626956311784 False
7 [ 1.33064721 -1.12499685  0.29465324 -0.06335068] [ 0.44721937 -0.7373708 ] -0.15628693521022796 False
8 [ 1.31776456 -1.14046893  0.29465324 -0.06335068] [-0.5014601   0.30409884] 0.48457166764265996 False
9 [ 1.30802781 -1.17436438  0.25853589  0.17118117] [-0.332757  0.676873] 0.3667040188678982 False
```

matteolucchi, 2021-12-23:
Hi @Rowing0914,

Thank you very much for using robo-gym and sorry for the late reply! I only had time now to look into the issue and some changes in the gym package caused the environments to break. 
It should be fixed now, if you are stil interested please  do a fresh installation and try it out and feel free to reopen the issue in case it is still not working. 

Cheers, 

Matteo

closed 2021-12-23 by matteolucchi