jr-robotics/robo-gym #52: Problems Starting a Simulated Robot Server

irivers29, 2021-12-08:
Hi,
first of all thank you for creating this amazing tool. If I get it running, I am sure that it will help me a lot with my research.

I am trying to install everything to run some experiments, but I am running into a few problems and running out of ideas on how to solve them. I am working on Ubunut 18.04 and I have created a conda environment with ros melodic and python 3.7. Once I clone every repositiory and follow the steps to install robo-gym and robo-gym-robot-servers I try to launch the mir robot server with following code:

`roslaunch mir100_robot_server sim_robot_server.launch gui:=true`

The output for this code is the following:

`RLException: [sim_robot_server.launch] is neither a launch file in package [mir100_robot_server] nor is [mir100_robot_server] a launch file name
The traceback for the exception was written to the log file
`

Also, if I try to run the MIR experiment, rviz opens the simulation, the server manager starts, but it fails to start a new robot server:

`Starting new Robot Server | Tentative 1 of 10` until it tries it 10 times

I woud really appreciate if someone could help me out with this issue.
Thank you

matteolucchi, 2021-12-23:
Hi @irivers29 and thank you for trying out robo-gym!

Sorry for the late reply. 

> `RLException: [sim_robot_server.launch] is neither a launch file in package [mir100_robot_server] nor is [mir100_robot_server] a launch file name The traceback for the exception was written to the log file`

This looks like the the package is not correctly sourced in your ROS workspace. Maybe something went wrong during the installation. Try to source the ROS workspace in the terminal where you run the roslaunch command with `source $ROBOGYM_WS/devel/setup.bash` where `$ROBOGYM_WS` is `~/robogym_ws ` if you followed the default installation instructions. 

Hope this helps, Cheers, 

Matteo

irivers29, 2021-12-23:
Hi @matteolucchi! Thanks for your reply! Luckily, I made it work a few days ago. Sorry for not updating the issue.
Thank you,
Ignacio

closed 2021-12-23 by irivers29