jr-robotics/robo-gym #55: Robot Server won't start

hayday100, 2022-02-19:
After installation, I ran:

python docs/examples/stable-baselines/td3_script.py

Then it takes forever to try to start the Robot Server:
Starting new Robot Server | Tentative 1 of 10
Starting new Robot Server | Tentative 2 of 10
Starting new Robot Server | Tentative 3 of 10
Starting new Robot Server | Tentative 4 of 10
Starting new Robot Server | Tentative 5 of 10
...

hayday100, 2022-02-21:
I got it to spawn Rviz and robot model.  Now it spawns Rviz too many times.  So I will move on ...

brean, 2022-04-02:
Hi @hayday100 

I just stumbled over this repo and installation was pretty easy and straight forward for me. It seems your installation was a success!

I am working with the MiR-100 simulation for quite some time now, you can ignore the error messages, they are known issues of the MiR-100-simulation.

You do not need a real robot!

These are my installation steps:
1. follow the installation instructions for noetic here: https://github.com/jr-robotics/robo-gym-robot-servers#installation. Note that the last step in the installation writes something in your bashrc-file, so after the last step close your terminal and open a new one to have a bash-instance that loaded this.
2. install robotgym using pip as described here: https://github.com/jr-robotics/robo-gym#installation
3. copy the code from this random-robot example as robot_random.py https://github.com/jr-robotics/robo-gym#random-agent-mir100-simulation-environment 
4. just run robot_random.py and wait a moment, rviz will open, the robot will spawn and after around 10 seconds the mir-robot will start moving.

Rasoul-Zahedifar, 2023-08-22:
Hello there,

I am writing this as I faced with this error and could solve it. Hopefully, it works for you. If you do NOT get any error during installation, but still have this issue, the problem is in sourcing in bashrc. Look at the line below which is from installing robo-gym-robot-servers:

printf "source /opt/ros/$ROS_DISTRO/setup.bash\nsource $ROBOGYM_WS/devel/setup.bash" >> ~/.bashrc

here, the path is $ROBOGYM_WS/devel/setup.bash, while it is highly likely that you installed it in another directory. Therefor, you should change the address accordingly. To check if it is changed correctly, run the command "tail .bashrc" and you should see the last line of the response being like "source correct path". Otherwise, it is highly likely that you did not install the packages correctly (in which case you should get error during installation). If this is the case and you get error in the command "catkin build", it might be due to the NOT existence of .catkin_tools along with src file. It is a hidden file and you can see hidden files with "ls -a". run the command "ls -a" in /home (or ~) directory and if you see .catkin_tools there, you may need to move it to $ROBOGYM_WS directory or bring src to /home directory. In the second case, make sure to change the source path I talked about above, accordingly.

Good Luck