jr-robotics/robo-gym #61: Error occurred while calling the reset function. Restarting Robot server ...

limeirax001, 2022-06-23:
Hello.
I tried to run the basic example "Random Agent MiR100 Simulation Environment" but the error appears "Error occurred while calling the reset function. Restarting Robot server ..."

Analyzing the tmux session it is possible to observe the following:

[more logs]
[INFO] [1656020731.484486, 11.575000]: Initializing robot_server node
[INFO] [1656020731.531181, 11.624000]: MiR 100 Sim Robot Server started at 47971
[ERROR] [1656020809.957213027, 147.568000000]: Updating ModelState: model [obstacle_0] does not exist
[ERROR] [1656020809.964521343, 147.576000000]: Updating ModelState: model [obstacle_1] does not exist
[ERROR] [1656020809.971933051, 147.585000000]: Updating ModelState: model [obstacle_2] does not exist

[exited]

then tmex is terminated.

------
I'm using ROS noetic on ubuntu 20.04.
Could you help me to solve this problem?

Adarsh3559, 2022-08-25:
I am facing the similar issue with Universal robot environment. 

Error occurred while calling the reset function. Restarting Robot server ...

Did you solved the error?

arehman1806, 2023-01-30:
I guess the latest bugfixes have not been published on pip. |You might be able to make it work by cloning the repo and installing the robo-gym package manually. This is how i got it to work.

Adarsh3559, 2023-01-30:
Hii thanks for the reply i already made it work in simulation but i was not
able to make it qork with real robot.

Alvin-shao, 2023-07-26:
> Hello. I tried to run the basic example "Random Agent MiR100 Simulation Environment" but the error appears "Error occurred while calling the reset function. Restarting Robot server ..."
>
> Analyzing [...]

Hello, I have been experiencing this error all the time. The reset function call has encountered an error. Have you resolved it？？

Alvin-shao, 2023-07-27:
> I am facing the similar issue with Universal robot environment.
> 
> Error occurred while calling the reset function. Restarting Robot server ...
> 
> Did you solved the error?

I am facing the similar issue with Universal robot environment.

Error occurred while calling the reset function. Restarting Robot server ...

Did you solved the error?20230727

Alvin-shao, 2023-07-28:
> 我在通用机器人环境中面临类似的问题。
> 
> 调用重置函数时出错。重新启动机器人服务器...
> 
> 你解决了错误吗？

Can you exchange ideas? My email is yanpeng_shao@163.com

Rasoul-Zahedifar, 2023-08-22:
Hello there,

I am writing for this issue as I faced with it too, but could solve the problem. hopefully, it works for you. If you can run gym.make properly and get the message "Successfully Started Robot Server", but get error while calling env.reset, this is because you installed robogym correctly but the problem comes from gym.make (or specifically, comes from gym, not robogym). You should install the gym version which is compatible with robogym. It is highly likely that you are using a newer version of gym. You may need to change to proper version by "pip install gym==0.18.3".

Good Luck