jr-robotics/robo-gym #65: 'NoneType' object has no attribute 'close'

Plusmile, 2022-09-05:
# error
Error occurred while calling the reset function. Restarting Robot server ...
Killing Robot Server at 127.0.0.1:46037 | Tentative 1 of 10

# my log:
rospy.internal][INFO] 2022-09-05 20:21:30,384: topic[/wrist_1_collision] adding connection to [http://ubuntu:45017/], count 0
[rospy.internal][INFO] 2022-09-05 20:21:30,387: topic[/tf_static] adding connection to [http://ubuntu:39745/], count 2
[rospy.internal][INFO] 2022-09-05 20:21:30,387: topic[/wrist_2_collision] adding connection to [http://ubuntu:45017/], count 0
[rospy.internal][INFO] 2022-09-05 20:21:30,392: topic[/tf_static] adding connection to [/robot_server], count 0
[rosout][INFO] 2022-09-05 20:21:30,391: ur3 Sim Robot Server started at 46037
[rospy.internal][INFO] 2022-09-05 20:21:30,470: topic[/move_objects] adding connection to [/objects_controller], count 0
[rospy.core][INFO] 2022-09-05 20:21:40,787: signal_shutdown [signal-2]
[rospy.internal][INFO] 2022-09-05 20:21:40,788: topic[/move_objects] removing connection to /objects_controller
[rospy.internal][INFO] 2022-09-05 20:21:40,789: topic[/tf_static] removing connection to http://ubuntu:37091/
[rospy.internal][INFO] 2022-09-05 20:21:40,791: topic[/rosout] removing connection to /rosout
[rospy.internal][INFO] 2022-09-05 20:21:40,791: topic[/env_arm_command] removing connection to /joint_trajectory_command_handler
[rospy.internal][INFO] 2022-09-05 20:21:40,791: topic[/tf_static] removing connection to /robot_server
[rospy.internal][INFO] 2022-09-05 20:21:40,792: topic[/clock] removing connection to http://ubuntu:45017/
[rospy.internal][INFO] 2022-09-05 20:21:40,792: topic[/tf_static] removing connection to http://ubuntu:39745/
[rospy.internal][INFO] 2022-09-05 20:21:40,792: topic[/clock] removing connection to http://ubuntu:45017/
[rospy.internal][INFO] 2022-09-05 20:21:40,793: topic[/joint_states] removing connection to http://ubuntu:45017/
[rospy.internal][INFO] 2022-09-05 20:21:40,793: topic[/tf] removing connection to http://ubuntu:36917/
[rospy.internal][INFO] 2022-09-05 20:21:40,793: topic[/tf] removing connection to http://ubuntu:36917/
[rospy.topics][ERROR] 2022-09-05 20:21:40,793: Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 326, in close
    c.close()
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 858, in close
    self.socket.close()
AttributeError: 'NoneType' object has no attribute 'close'

[rospy.internal][INFO] 2022-09-05 20:21:40,794: topic[/tf_static] removing connection to http://ubuntu:36917/
[rospy.internal][INFO] 2022-09-05 20:21:40,794: topic[/tf_static] removing connection to http://ubuntu:36917/
[rospy.topics][ERROR] 2022-09-05 20:21:40,794: Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/topics.py", line 326, in close
    c.close()
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 858, in close
    self.socket.close()
AttributeError: 'NoneType' object has no attribute 'close'

[rospy.internal][INFO] 2022-09-05 20:21:40,794: topic[/shoulder_collision] removing connection to http://ubuntu:45017/
[rospy.internal][INFO] 2022-09-05 20:21:40,795: topic[/upper_arm_collision] removing connection to http://ubuntu:45017/
[rospy.internal][INFO] 2022-09-05 20:21:40,795: topic[/upper_arm_collision] removing connection to http://ubuntu:45017/
[rospy.internal][INFO] 2022-09-05 20:21:40,795: topic[/forearm_collision] removing connection to http://ubuntu:45017/
[rospy.internal][INFO] 2022-09-05 20:21:40,795: topic[/forearm_collision] removing connection to http://ubuntu:45017/

# my version
PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.14
ubuntu 20.04

# my codeimport gym
import robo_gym
from robo_gym.wrappers.exception_handling import ExceptionHandling

target_machine_ip = '127.0.0.1' # or other machine 'xxx.xxx.xxx.xxx'

env = gym.make('EndEffectorPositioningURSim-v0', ur_model='ur3',ip=target_machine_ip, gui=True)
env = ExceptionHandling(env)

num_episodes = 10

for episode in range(num_episodes):
    done = False
    env.reset()
    while not done:
        state, reward, done, info = env.step(env.action_space.sample())

Plusmile, 2022-09-05:
i try to use this solution，but failed.
[Solution](https://github.com/ros/ros_comm/issues/2212)

Plusmile, 2022-09-05:
# version 
proboc 3.6.1
probobuf 3.20.1

dear  Matteo,thank you very much

Plusmile, 2022-09-06:
I After painful repeated uninstall and re-execution of the instructions, I solved the problem. I initially determined that it is the problem of apt changing the source. Do not change the source in this project, especially when you are in China

closed 2022-09-06 by Plusmile

Alvin-shao, 2023-07-27:
> 经过痛苦的反复卸载并重新执行指令，我解决了问题。我最初确定这是适当更改源的问题。不要更改此项目中的源，尤其是当您在中国时

我也是这个问题，请问您是怎么解决的呢？

Alvin-shao, 2023-07-27:
> 经过痛苦的反复卸载并重新执行指令，我解决了问题。我最初确定这是适当更改源的问题。不要更改此项目中的源，尤其是当您在中国时

Hello, I have also encountered the same problem as you. I am also in China and I have been unable to solve this problem. Other installations have not had any problems, except for resetting function calls. Can you share how you resolved it?