jr-robotics/robo-gym #70: Connection issue with real UR5

kirstyellis, 2022-10-26:
Hi,

We are trying to run robo-gym with a real UR5. We successfully ran the simulation beforehand, but after trying to run the gym with a real robot, we are hitting an error. This is the code that we are running:

`import gym
import robo_gym
from robo_gym.wrappers.exception_handling import ExceptionHandling

# initialize environment
env = gym.make('EndEffectorPositioningURRob-v0', ip='127.0.0.1:50051')
env = ExceptionHandling(env)

num_episodes = 10

for episode in range(num_episodes):
    print("start episode")
    done = False
    env.reset()
    while not done:
        # random step in the environment
        state, reward, done, info = env.step(env.action_space.sample())
`
This is the error that we are getting:

env.reset()
  File "/home/kirstyellis/.local/lib/python3.8/site-packages/gym/wrappers/order_enforcing.py", line 18, in reset
    return self.env.reset(**kwargs)
  File "/home/kirstyellis/robogym_ws/src/robo-gym/robo_gym/envs/ur/ur_base_env.py", line 104, in reset
    if not self.client.set_state_msg(state_msg):
  File "/home/kirstyellis/.local/lib/python3.8/site-packages/robo_gym_server_modules/robot_server/client.py", line 17, in set_state_msg
    msg = self.robot_server_stub.SetState(state_msg, timeout = 60)
  File "/home/kirstyellis/.local/lib/python3.8/site-packages/grpc/_channel.py", line 946, in __call__
    return _end_unary_response_blocking(state, call, False, None)
  File "/home/kirstyellis/.local/lib/python3.8/site-packages/grpc/_channel.py", line 849, in _end_unary_response_blocking
    raise _InactiveRpcError(state)
grpc._channel._InactiveRpcError: <_InactiveRpcError of RPC that terminated with:
	status = StatusCode.DEADLINE_EXCEEDED
	details = "Deadline Exceeded"
	debug_error_string = "UNKNOWN:Deadline Exceeded {created_time:"2022-10-26T11:29:40.308375201-04:00", grpc_status:4}"

We entered the rs_address as 127.0.0.1 and the port given by the robot server. We found that if we entered a completely random port number, we had the same output. Is there a way to determine if a connection is made with the robot server? 
The ur5 arm is launching correctly, we can see that in rviz the joint states are being read so I don't think it is an issue with the robot side of things.

If you could provide any suggestions as to how we can get around this issue, that would be great. 
Thanks!

Adarsh3559, 2022-10-27:
Hii @kirstyellis , I am getting the exact same error, everything works fine in simulation but when trying with real UR10e I am getting the same error, Please also let me know if you come up with the solution. I would highly appreciate that.

kirstyellis, 2022-10-27:
hi @Adarsh3559  I saw your issue, the error is slightly different to the one I am seeing. 

I can reproduce your error if I change the  rs_address IP to an incorrect IP. 

I have my robot server running on my local machine and so am using ip:port as 127.0.0.1:50051. I changed the IP to an arbitrary value and then got the exact error that you have reported.

kirstyellis, 2022-10-27:
@Adarsh3559 I found the problem on our setup. The controller that the joint_trajectory_command_handler.py was publishing to was different to the controller that we had loaded. By default the controller that is loaded by the ur bringup launch is scaled_pos_joint_traj_controller. The publisher in robo-gym servers was publishing to pos_traj_controller.

See this line: https://github.com/jr-robotics/robo-gym-robot-servers/blob/master/ur_robot_server/scripts/joint_trajectory_command_handler.py#L16

Adarsh3559, 2022-10-28:
@kirstyellis , Can you help me out also, as i am new to ROS, i am also running the robot server on my local machine, but i have connected my robot via Ethernet cable and the "**robot IP is static : 192.168.1.70 with  subnet mask: 255.255.255.0**" while i have changed my "**Ubuntu IP to 192.168.1.4 with subnet mask:255.255.255.0**" otherwise when i ping the robot its says host unreachable if i don't change IP on Ubuntu.

And then i start the robot server and bring up the UR10e robot i can also he UR10e arm is launching correctly, I can also see that in rviz the joint states are also being read as you mentioned above.

**I am running the below code.

#%%
import gym
import robo_gym

 # initialize environment
env = gym.make('BasicAvoidanceURRob-v0', ur_model='ur10e', rs_address='192.168.1.4:50051')

#%%
num_episodes = 10

for episode in range(num_episodes):
    done = False
    env.reset()
    while not done:
        # random step in the environment
        state, reward, done, info = env.step(env.action_space.sample())
#%%**

Can you tell me what changes i need to make in order to make it work with my UR10e. I would highly appreciate your help,
Also,  if you want we can talk via email **My email is : adarsh3559@gmail.com**.  Please help me out as i am stuck on this problem for too long

kirstyellis, 2022-10-28:
I would try rs_address='127.0.0.1:50051'

Adarsh3559, 2022-10-28:
@kirstyellis  I tried but same error .do i need to make changes in https://github.com/jr-robotics/robo-gym-robot-servers/blob/master/ur_robot_server/scripts/joint_trajectory_command_handler.py#L16 ?

kirstyellis, 2022-10-28:
Is the error the same as you had before or the same as the error I had? The error you reported in the issue that you opened is different to the one I reported. 

You will need to change that line in the handler if the name of the controller you are using has a different name. You can check the name of the controllers that you have by making a rosservice call to list_controllers. Then check if the name of your controller matches the one given in the link.

KarimHP, 2023-03-31:
> hat the joint_trajectory_command_handler.py was publishing to was different to the controller that we had loaded. By default the controller t
@kirstyellis 

So im facing the same issue. Could you elaborate what you changed? Did you change the ur5_bringup.launch file from the UR_Robot_driver? or did you change the line the script you are referring to? I know its been a while since u posted but i just saw it and i have the exact same problem so i would be really really grateful for any help!

kirstyellis, 2023-04-18:
@KarimHP, which controller is being loaded?