jr-robotics/robo-gym #71: I cant get robogym to run the examples, it has TypeError: Descriptors cannot not be created directly.

zonared, 2023-02-15:
Hi there, I am having an issue with running the example using this command:

user@computer:~/robogym_ws$ roslaunch mir100_robot_server sim_robot_server.launch gui:=true

I'm receiving this error:

Traceback (most recent call last):
  File "/home/user/robogym_ws/devel/lib/mir100_robot_server/robot_server.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/user/robogym_ws/src/robo-gym-robot-servers/mir100_robot_server/scripts/robot_server.py", line 6, in <module>
    from mir100_robot_server.ros_bridge import RosBridge
  File "/home/user/robogym_ws/src/robo-gym-robot-servers/mir100_robot_server/src/mir100_robot_server/ros_bridge.py", line 16, in <module>
    from robo_gym_server_modules.robot_server.grpc_msgs.python import robot_server_pb2
  File "/home/user/.local/lib/python3.8/site-packages/robo_gym_server_modules/robot_server/grpc_msgs/python/robot_server_pb2.py", line 36, in <module>
    _descriptor.FieldDescriptor(
  File "/home/user/.local/lib/python3.8/site-packages/google/protobuf/descriptor.py", line 560, in __new__
    _message.Message._CheckCalledFromGeneratedFile()
TypeError: Descriptors cannot not be created directly.
If this call came from a _pb2.py file, your generated code is out of date and must be regenerated with protoc >= 3.19.0.
If you cannot immediately regenerate your protos, some other possible workarounds are:
 1. Downgrade the protobuf package to 3.20.x or lower.
 2. Set PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION=python (but this will use pure-Python parsing and will be much slower).

More information: https://developers.google.com/protocol-buffers/docs/news/2022-05-06#python-updates

My protoc --version is libprotoc 3.6.1
and pip install 'protobuf<=3.20.1' --force-reinstall, which installs v3.20.1
and Python 3.8.10

What should I do?
I'm running a fresh install of Ubuntu 20.04 and ROS Noetic full desktop
the catkin build completes without issue. I also receive heaps of warnings like "F_REPEATED_DATA ignoring data with redundant timestamp for frame" but I figure I need to fix the error first.

Thanks in advance, Zonared

Adarsh3559, 2023-02-15:
pip install  protobuf==3.20, I would suggest you to look at open issues and discusions.

Rasoul-Zahedifar, 2023-08-22:
Hello there,

I am writing this a I faced with this problem and could solve it. You need to enter "pip install protobuf==3.20.0" and procced.

Good Luck