jr-robotics/robo-gym #75: Error occurred while calling the reset function. Restarting Robot server ...

Alvin-shao, 2023-07-26:
Error occurred while calling the reset function. Restarting Robot server ...
Killing Robot Server at 127.0.0.1:35015 | Tentative 1 of 10
Successfully killed Robot Server at 127.0.0.1:35015
Starting new Robot Server | Tentative 1 of 10
Successfully started Robot Server at 127.0.0.1:44493
Error occurred while calling the reset function. Restarting Robot server ...
Killing Robot Server at 127.0.0.1:44493 | Tentative 1 of 10
Successfully killed Robot Server at 127.0.0.1:44493
Starting new Robot Server | Tentative 1 of 10
Successfully started Robot Server at 127.0.0.1:42563
Error occurred while calling the reset function. Restarting Robot server ...
Killing Robot Server at 127.0.0.1:42563 | Tentative 1 of 10
Successfully killed Robot Server at 127.0.0.1:42563
Starting new Robot Server | Tentative 1 of 10

I am using Ubuntu 20.04，robo-gym==V1.0.0, I see there was a problem before that was the same as mine, but after I downgraded the version, the UR5 robotic arm no longer works. May I ask how to solve it?

Alvin-shao, 2023-07-26:
@matteolucchi 
Can you help to see how this situation can be resolved? Thank you!

Alvin-shao, 2023-07-26:
@matteolucchi 

Can you spare some time from your busy schedule to take a look at this issue? Thank you!

closed 2023-08-01 by Alvin-shao

Adarsh3559, 2023-08-01:
Hii,

You can try reinstalling everything, also downgrade your protobuf package below 20.04 or something like that, I think then it might work. For me also i had to do so many trial and error. it worked in simulation. However was not able to make it work with real system. Also i would like to mention that when downloading Universal robots driver i used this repo insted of going with Robogym. When you reinstall everything, build your workspace from below link. I was using ROS noetic with Ubuntu 20.04

I think then you might be able to make it work in simulation.

https://github.com/UniversalRobots/Universal_Robots_ROS_Driver

I hope it helps.

Best Regards,
Adarsh Sharma