jr-robotics/robo-gym #76: An error occurred when calling the reset function. Emergency！！

Alvin-shao, 2023-07-27:
Hello, I used Ubuntu 18.04 a few days ago, and now I have installed an Ubuntu 20.04 system. However, the problems that occur on these two systems are the same, both of which are calling reset function errors.
[image]
I have no issues starting the service manager with commands on the terminal.

[image]
There are no issues with installing robot gym, and there are no issues with compiling the workspace of the robot. I'm not sure where the problem lies. Can you help solve it? Thank you！！

Alvin-shao, 2023-07-27:
@matteolucchi 
Can you help solve it? Thank you！！

Alvin-shao, 2023-07-28:
在log里找到了这个错误，怎么解决？
[image]

Alvin-shao, 2023-08-02:
问题已解决，有需要知道怎么解决的可以加我qq201976802

Rasoul-Zahedifar, 2023-08-22:
Hello there,

I am writing for this issue as I faced with it too, but could solve the problem. hopefully, it works for you. If you can run gym.make properly and get the message "Successfully Started Robot Server", but get error while calling env.reset, this is because you installed robogym correctly but the problem comes from gym.make (or specifically, comes from gym, not robogym). You should install the gym version which is compatible with robogym.  It is highly likely that you are using a newer version of gym. You may need to change to proper version by "pip install gym==0.18.3".

Good Luck

Alvin-shao, 2023-09-04:
> 嗨，你好
> 
> 我也面临着这个问题，我正在为这个问题写作，但可以解决问题。希望它对您有用。如果您可以正确运行 gym.make 并收到消息“已成功启动机器人服务器”，但在调用 env.rereset 时出现错误，这是因为您正确安装了 robogym，但问题来自 gym.make（或者具体来说，来自健身房，而不是 robogym）。您应该安装与机器人健身房兼容的健身房版本。您很可能使用的是较新版本的健身房。您可能需要通过“pip install gym==0.18.3”更改为正确的版本。
> 
> 祝你好运

Thank you very much for your answer. I tried it and it was indeed a problem with the GYM version. It can now be fully run.
Good luck！

jr-b-reiterer, 2024-07-17:
Clarification on gym version compatibility planned in the course of #73.

closed 2024-07-17 by jr-b-reiterer